dhalperi/linux-80211n-csitool-supplementary #244: Getting CSI from a sensor node

BelalKorany, 2016-10-14:
Hi All, 

I want to get the CSI measurements but from many different locations, so connecting the Intel Card to many laptops and putting them in those locations would be impractical. 

I was thinking of attaching the card to a board that has the PCIe interface (like Intel Galileo Gen 2). I think the board also runs on Linux which makes it more suitable for the CSI tool. There has been a previous discussion on this but no conclusion was drawn. I am new to this area so I was wondering if anybody has tried that before or has any thought of how to make it work?

If it is not supposed to work, any other suggestions for such an application? 

Thanks.

tahmidzbr, 2016-10-15:
@BelalKorany  Hi there, I also want to do something like that, for my research work. Because having too many laptops would be impractical as well, i have some ideas that we can try but not sure, do you want to collaborate on how to atleast get it working? and once it starts working then we go our own ways and do our personal research? let me know what you think!
Tahmid

ppenumarthi, 2017-11-15:
Do you guys have any conclusion? I am planning to use Galileo 2nd Gen and am clueless what to predict.