dhalperi/linux-80211n-csitool-supplementary #246: How to get ID number or source address in CSI packet?

ccjack, 2016-10-19:
Hi, 
Is there any way to know the ID information or source address regarding the receiving packet's CSI? I have set up a linux-computer in AP-mode where there are 4 smartphone are connected. Then, the AP has receive a sequence of CSI from the four smartphone beacons even without pining them. However, it is hard to know each CSI belong to which smartphone. I have checked the cn_msg->id.val (in connector.h) which has always the same ID number even regarding different clients. Is there any way that I can know the present CSI belong to which clients (i.e smartphone)? Otherwise, I may need to use wireshark to map each CSI data with incoming packets. Any comments and suggestion will be appreciated! Thanks!

ccjack, 2016-11-01:
I figure out how to do it. Thanks!

closed 2016-11-01 by ccjack

asifhanif1122, 2016-11-02:
@ccjack Would you like to share method for identifying ID of clients?

swashah, 2016-11-22:
@ccjack :Can you please briefly guide how can use sequence number of packets. Currently I have 02 laptops in monitor mode and third laptop is injecting packets. Now logically if 02 antennas are close enough then we will have similar CSI-amplitude. I am getting similar values but there is phase shift so can you please guide how can we use the sequence number to align signals?

tahmidzbr, 2017-01-07:
@swashah Hi did you figure out how to get the sequence number of packets? If so can you please explain how. Many thanks.

swashah, 2017-01-07:
@tahmidzbr : no i did'nt

rednom, 2017-04-12:
@swashah @tahmidzbr Hi did you get how to get sequence number??