dhalperi/linux-80211n-csitool-supplementary #247: How to change power level in Intel 5300 WLAN card?

asifhanif1122, 2016-10-20:
Hi Everyone,
After installing modified firmware , commands to change power level in WLAN card don't work. @dpward @dhalperi  Are changes in power level disabled in modified firmware? I have used all possible commands available on different forums to change power level but they did not work. Power level shown by _**iwconfi**_g command is always 15dBm. I want to lower it.

soltanaghaei, 2016-11-04:
@asifhanif1122 I have the same issue for lowering the transmission power. Did you find any answer for your question?

asifhanif1122, 2016-11-05:
I am still looking for solution. If I get it, I will post here. Meanwhile, if you get successful to find out, do let me know.

tahmidzbr, 2016-11-09:
@asifhanif1122 @soltanaghaei Hi im not sure if i make any sense but are you guys referring to the transmit power? I.e if you use a router as transmitter than i guess there must be a way to control the power of router? Thing is with some routers there is auto power control and you can set it like 25%,50%,100% and even 0%. My router doesnt have that option, but i would rather want one where i can set the auto power control to 0%. Do you by any chance know if i can do that using the command line in Ubuntu?

soliloquy2017, 2017-03-13:
I have faced the same problem as how to change the tx power in Intel 5300 WLAN card. Have you ever resolved this problem? @asifhanif1122 @soltanaghaei

DontKnowMy, 2017-04-05:
I am looking for the same answer. Who has it? @asifhanif1122 , @soltanaghaei , @soliloquy2017 , @dhalperi

mikeWangJC, 2017-05-02:
i tried some commands but failed

chukku, 2018-06-20:
I am also looking for a way to find out if my router is doing automatic power control or not. I need a router where there is no automatic power control. Any suggestions ?