dhalperi/linux-80211n-csitool-supplementary #249: Why can't I get CSI at sometimes?

chensiwen, 2016-10-26:
I have installed the CSI tool correctly and could get CSI data many times, but I can not get CSI data at sometimes. Why?

And I install it again, but the result  is as the attach file.
Specially, I think the problem occurs firstly at the Step2 as follows:
chensiwen@chensiwen:~/linux-80211n-csitool$ git checkout ${CSITOOL_KERNEL_TAG}
**error: pathspec 'csitool-4.4' did not match any file(s) known to git.**
What is the 'csitool-4.4'?

[image]
[image]

Please help me! 
Thank you very much!

chensiwen, 2016-11-02:
The problem is done. My Linux kernel is 4.4, so I change it to 4.2.

closed 2016-11-02 by chensiwen

arpapad, 2017-12-01:
Hi, how did you change your Linux Kernel?