dhalperi/linux-80211n-csitool-supplementary #251: process CSI with C or C++?

linkzhao, 2016-11-06:
Hi all,
I am a new guy to this area , I want to konw  how  process CSI with C or C++? dose anyone do  it?
thanks!!

brunosoaresds, 2016-11-15:
@linkzhao When i initiate in this area i have tried to process the data with C/C++ program too, but i strongly recommend you to learn how matlab works and work with it, because in matlab the most part of functions what you need is already implemented.

But, if you want to process the data in c/c++ you will need to study how this functions works:
- abs() - for complex numbers = sqrt(pow(real_part,2) + pow(imag_part,2))
-- This function will give to you the magnitude (amplitude) of some CSI trace
- db() - This function will convert the magnitude of some trace to decibel (SNR in dbm)
- get_scaled_csi() - This function will normalize the CSI trace, removing AGC gain and thermal noises of CSI trace

linkzhao, 2016-11-16:
thanks for your reply！but if can not  process the data in real time ,how to solve the problem of localization in real time ?

linkzhao, 2016-11-16:
thank you for your reply, but if can not process the data with C or C++,how to solve the problem of localization in real time ?

brunosoaresds, 2016-11-16:
@linkzhao I don't said that is not possible to process the data with C/C++, i just said that is so much easy to process with matlab because the most part of functions what you will need is already implemented and for your research you can start processing data not in real time and after you can develop a program that proccess in real time.

However, if you want to start with a C/C++ program and real time system you should create all functions again, and like i explain before you can start with amplitude processing what is made by basically this three functions: db(), abs() and get_scaled_csi()

linkzhao, 2016-11-16:
got it！thanks，I have another question ,Are you plot CSI with time,Why there is an uncertain delay at the start of measured CIR samples?

linkzhao, 2016-11-16:
@brunosoaresds like this:
[image]

linkzhao, 2016-11-22:
@brunosoaresds hi,can you try to link wifi by using "iwconfig"?