dhalperi/linux-80211n-csitool-supplementary #253: What's the difference between different transmit rate in inject-monitor mode?

hot-dog, 2016-11-09:
In inject-monitor mode, we need set transmit rate by "sudo echo 0x4101 | sudo tee 'find /sys -name monitor_tx_rate'", i am wondering what's the difference between different transmit rate. Can anyone tell me something about that? Thanks!

swashah, 2016-11-09:
@hot-dog : are you able to set up injection-monitor mode successfully? I am trying but CSI is not logged on receiver side? Any idea?

hot-dog, 2016-11-10:
Yes, I can. The original setup shell scripts did not work for me, I modified it and it worked. Use iwconfig command to make sure you have setup injection-monitor mode correctly, for transmitter it will show "managed", for receiver it will show "monitor", if not, you may have to modified the original shell scripts files to fit yourself.

swashah, 2016-11-10:
@hot-dog : thanks for you suggestion. Can I please have the shell script you used? My email is wajidnaqvi2000@yahoo.com.
For your question regarding you can refer to issue #94 which describes it in detail.
Cheers

swashah, 2016-11-10:
@hot-dog  i mean you can refer issue # 94 regarding transmit rate

hot-dog, 2016-11-10:
@swashah sure, I have sent the shell script to your mailbox, please check it. I know how to analyze the rate format, what i want to know is the influence of different transmit rate on CSI measurement.

swashah, 2016-11-10:
@hot-dog : whenever i execute ./setup_inject .sh it gives FATAL: Module iwlwifi is in use or responds nothing. Can you please help me out. I also tried your shell script but its not working through. I am on Ubunt 12.04. More over I can collect csi if I connect to commercial AP and ping it. Many thanks

swashah, 2016-11-10:
@hot-dog : Issue resolved with slight modification... Thanks

hot-dog, 2016-11-11:
@swashah Glad to hear that!

swashah, 2016-11-11:
@hot-dog : do you have any idea which is the antenna A on NIC ? Is it one on port number 1?

hot-dog, 2016-11-11:
@swashah I guess so.

swashah, 2016-11-11:
@hot-dog : Thanks

swashah, 2016-11-14:
@hot-dog : do you have any idea how can one synchronize the time stamps of two device recording csi simultaneously in monitor mode while third device is injecting packets?

hot-dog, 2016-11-14:
@swashah  If just a rough synchronization, you don't have to do anything. But if you want achieve a very precise synchronization, say microsecond level or nanosecond level, i'm sorry, i have no idea too, the packet receiving and processing time is uncertain for every machine. Maybe you can search for other issues and you might find something.

swashah, 2016-11-14:
@hot-dog ; thanks

swashah, 2016-11-14:
@hot-dog : I am trying to collect CSI on 02 laptops very close to one another in same environment but getting different CSI amplitudes most of time (some time similar). I am assuming they should be very similar. Any input/suggestion from you side please?

hot-dog, 2016-11-15:
@swashah I think that is normal. If you are using 5G band, then the wave length is 6cm, if the distance of two antennas is greater than half of the wave length, i.e 3cm, then the recording CSI will be different. Actually, i have a similar problem, i have recorded several CSIs at different time in a stable indoor environment, theoretically they should be similar, but in fact quite different to each other, i am very puzzled about this.

swashah, 2016-11-15:
@hot-dog : Thanks for your input. I am operating at 2.4GHz means that half of wavelength is about 6 cm. Even if i keep distance less than 6cm i get some time different measurements but sometime similar as well. Also I have noticed that when the reading are similar they are not aligned. I am looking into this. Good luck for your problem. (plz let me know if you figure out some thing, I will keep you update)

hot-dog, 2016-11-15:
@swashah Sure, i will.

swashah, 2016-11-19:
@hot-dog : When I collect CSI on two laptops in monitor monitor mode their data is similar but not aligned. Do you have any suggestions how can we align them based upon time stamp, as I have seen a work they are doing it like this, but I am not getting hold of this. Also there is another slightly modified version of CSI (link I sent to your email) that inculcates frame sequence number but I could not setup that tool it gives error when binaries are generated.

swashah, 2016-11-29:
@hot-dog : did you figure out CSI dissimilarity issue. Actually I am trying to collect CSI while keeping distance between 02 antennas less than half wavelength, but not observing similar csi over all subcarrier. however few sub-carriers are similar as well...

hot-dog, 2016-12-05:
@swashah Is the difference large? If the difference is small, i think it's normal.

swashah, 2016-12-05:
@hot-dog : on some sub-carriers difference is quite large, I also assume that difference of up to 3 units (in amplitude) should be normal but difference of 7 or 8 in amplitude should not be there?

hot-dog, 2016-12-05:
@swashah Have you ever tried to collect data in the same location but at different time? Is these data similar?

swashah, 2016-12-05:
@hot-dog : yes if environment stays same, data comes up similar...same location means I have deployed external antenna and place them on same location...

hot-dog, 2016-12-05:
@swashah Does the two antennas connect to one NIC card? If so, then it is quite normal the data is different, if not, i guess the connecting line between antenna and NIC card cause the dissimilarity, such as its length, material......., after all, the connecting line is also part of the channel.

swashah, 2016-12-05:
@hot-dog : 02 antennas are connected to two different NIC on 02 laptops...You could be right regarding length of wire....Connecting wires of antenna are of same length but I guess the way they run from antenna to NIC (e.g. bends etc) are different I guess...

swashah, 2016-12-19:
@hot-dog : did you figure out how rate effect CSI in monitor/inject mode?

hot-dog, 2016-12-29:
@swashah I have tried different rates, it seems there is no big difference.

sapansanu, 2017-11-01:
@hot-dog @swashah I am having the same problem. I can send the packets but cannot receive it on the receiver. Can you guys send the modified script to me too? My email is sapansanu@gmail.com. Thanks!