dhalperi/linux-80211n-csitool-supplementary #254: the different about get_scaled_csi_sm.m and  get_scaled_csi.m?spatial mapping

linkzhao, 2016-11-11:
Hi all,
I am new to this area,could anyone tell me the different about get_scaled_csi_sm.m and  get_scaled_csi.m? and what spatial mapping is mean?
thanks!

tahmidzbr, 2016-11-21:
@linkzhao did you figure it out?

linkzhao, 2016-11-21:
sorry，i don't  figure it out so far,and please tell me when you konw the answer.thanks

tahmidzbr, 2016-11-21:
@linkzhao sure. Although can i ask you a question?
Did you try using a router as transmitter and laptop with 5300 as receiver? When i try pinging and logging CSI i see that for some packets my no. of transmit antennas and no. of receive antennas vary.
Like if i ping 100pkts/s for say 2-3 minutes some packets show Nt=2,Nr=3 some show Nt=1 Nr=2 and it keeps varying. Do you happen to know why?

linkzhao, 2016-11-21:
i have not meet such problem，maybe you should check your antenna  and try to change a router

lidongning, 2017-07-10:
I have the same question ,  Nt=2,Nr=3 ,but some packets show Nt=1 Nr=2 and it keeps varying.  could you help me ? @tahmidzbr

AliAwad1994, 2018-10-08:
you have to fix the transmission rate to solve the problem of varying Rx and Tx... it's hard to find a commercial AP that support this feature.