dhalperi/linux-80211n-csitool-supplementary #255: Using fixed hopping sequence in simple transmitters to measure one-way CSI

nmccouat, 2016-11-12:
Hi, cool project. I was wondering if you could use (very) simple transmitters programmed to do a fixed hopping sequence, and measure the CSI only at the receiver

It would be great to be able to track multiple mobile devices using very simple / compact / battery-powered wifi hardware (e.g. esp8266) and measure CSI / TOF on one IWL5300 card operating as a base station

Are the limitations on this the clock stability of the roving hardware? Presumably the hopping sequence itself is fairly simple and could simply be hard-coded to occur at regular intervals

awelkie, 2016-11-12:
How are you planning on doing the tracking? If you're planning on doing something like in the [Chronos](https://www.usenix.org/system/files/conference/nsdi16/nsdi16-paper-vasisht.pdf) paper, note that they measure the CSI at both the transmitter and receiver to overcome the phase ambiguity.

If you can't measure the CSI at your mobile nodes, then you could estimate the angle-of-arrival of the signal at the base station. That could at least tell you in what direction the mobile nodes are located.

nmccouat, 2016-11-14:
Ah, good one - thanks. Yes I was mostly interested in a single LOS measurement and whether it could be simplified for that case. 
I guess you could possibly do something similar with two receivers separated by a specific offset in that line of sight - so you can still do all the grunt work on the base station, and keep rovers simple
Ta anyway. More reading to do  :)