dhalperi/linux-80211n-csitool-supplementary #257: Data misalignment on two laptops

swashah, 2016-11-19:
Hi, I have two laptops in monitor mode and third laptop is injecting packets. Two laptops are very close to each other and when I analyze the CSI amplitude in matlab data seems to be similar (as it should be logically) misalligned. Is there any way to align the data measured by two laptops based upon time stamp or any thing else. Thanks

awelkie, 2016-11-20:
What do you mean by "misaligned"? Could you provide some plots or data demonstrating this misalignment?

swashah, 2016-11-20:
@awelkie : I have attached a figure,as you can see the data is similar but there is no sort of synchronization?
[image]

awelkie, 2016-11-20:
Are you plotting amplitude or just the real value? And is the x-axis subcarrier?

swashah, 2016-11-20:
@awelkie : I am plotting abs(csi) i.e. amplitude and x-axis is number of packets. Graph is for single sub-carrier no. 14. so these are variations of sub-carrier 14 for 45 injected packets.

swashah, 2016-11-20:
@awelkie : any idea or suggestion> Did you get what I mean?

awelkie, 2016-11-20:
Yeah, I understand what you mean by misalignment now, but unfortunately I don't have any suggestions. I thought you might be talking about differences between subcarriers, but that's not the case.

swashah, 2016-11-20:
@awelkie : No worries , thanks for your time

hot-dog, 2016-11-22:
@swashah Is the delay(packet number between two peaks) fixed? If it is, then the misalignment should not be a matter. I think this misalignment is caused by one monitor ignores some packets in the beginning, it's just a guess, i don't know the true reason either.

swashah, 2016-11-22:
@hot-dog : I can't get what do you mean by 'delay fixed'. Can you elaborate a bit?

hot-dog, 2016-11-22:
@swashah I mean the gaps described by the green line.
[image]

swashah, 2016-11-22:
@hot-dog : Apparently they are, but I have not checked in matlab. I will do in the morning and check this out....