dhalperi/linux-80211n-csitool-supplementary #261: someCSI

luxiangx, 2016-11-30:


swashah, 2016-11-30:
@luxiangx : check out the antenna connections for corresponding values. May be antennas are not plugged properly...

luxiangx, 2016-12-01:
@swashah :Thank you for your answer. But the 0 values are few, it may not be the cause of the antenna.

swashah, 2016-12-01:
@luxiangx : is it the same antenna which shows 0?

luxiangx, 2016-12-01:
@swashah :different antennas and different sub-carriers

swashah, 2016-12-01:
@luxiangx : At same place?

luxiangx, 2016-12-01:
@swashah :yea,i just not sure these 0 values are normal,and can i use them?

swashah, 2016-12-01:
@luxiangx : try to take some readings at some other place, it will give you better insight...

luxiangx, 2016-12-01:
many thanks

hot-dog, 2016-12-05:
@luxiangx I also get some zero values, it seems normal.

tahmidzbr, 2016-12-05:
@luxiangx @swashah @hot-dog  do you know what could cause it to give some 0 values? It may seem normal, but is there any explanation to this?