dhalperi/linux-80211n-csitool-supplementary #264: Why the unwrapped CSI phase is nearly liner?

hot-dog, 2016-12-05:
Recently i am doing phase sanitization and there are some doubts that bothers me a lot. Why the unwrapped CSI phase is nearly liner?
[image]
The phase before unwrap operation is zigzag and in local part, it shows perfect linearity(figured out in red lines).
[image]
I know how the unwrap operation operated. Can anyone tell me the reason or give me some materials related? Thanks a lot!

caichao, 2016-12-16:
Hi, from my perspective view, the data presented in figure 1 is quite good. Can you post me some of your measurements for processing? 
To explain why this linearity exist, you can refer to the paper: SpotFi, decemeter-level indoor localization.

hot-dog, 2016-12-29:
@caichao It's someone else's data measurements, not mine.

sunyu1122, 2017-01-03:
@hot-dog Hello, I want to know how the unwrap operation operated,thanks

hot-dog, 2017-01-03:
@sunyu1122  Using MATLAB function unwrap.

sunyu1122, 2017-01-03:
@hot-dog thanks for your help

sunyu1122, 2017-01-03:
@hot-dog I visited your github home page and found that you are also learning the indoor location.are you read the paper "Precise Power Delay Profiling with Commodity WiFi"? If you do , i want to know how to obtian the initial time about the CIR.That is to say, how do you obtain Precise Power Delay Profiling? I sincerely hope to get your reply.

hot-dog, 2017-01-03:
@sunyu1122 Sorry, my research is not about indoor localization but related to it, and i have not read that paper.

sunyu1122, 2017-01-03:
@hot-dog thanks