dhalperi/linux-80211n-csitool-supplementary #265: 1 Computer connected to another Computer ( Access Point)

danha170, 2016-12-05:
Computer 1--> Ubuntu 11.09
Computer 2--> Ubuntu 14.04.01

How to connect these two PC's, as one computer becomes the AP and another a client. 
As I tried to make a **hotspot** on **one computer** and I **ping** it successfully but could not able to collect the **CSI data.**
So how could I successfully make these two computers connect wireless and can collect the **CSI data** on both at the same time after **ping** them at the same time.

danha170, 2016-12-08:
Computer 1--> Ubuntu 11.09
Computer 2--> Ubuntu 14.04.01

How to connect these two PC's, as one computer becomes the AP and another a client.
As I tried to make a hotspot on one computer and I ping it successfully but could not able to collect the CSI data.
So how could I successfully make these two computers connect wireless and can collect the CSI data on both at the same time after ping them at the same time.

sunyu1122, 2016-12-12:
Have you resolved this problem? I meet the same question. If you have resolved ,please give me a help,thanks

danha170, 2016-12-13:
Unfortunately I am still looking for a solution.

smallsmart1994, 2018-03-26:
Have you resolved this problem? I meet the same question. I am a graduate student at USTC and major in CSI. Are you Chinese？Can we add WeChat  or other communication software in order to discuss the details about this problem, thanks