dhalperi/linux-80211n-csitool-supplementary #268: MIMO system,how to know the each dimension of the CSI matrix which represent of single transmit−receive antenna pair

linkzhao, 2016-12-08:
A commercial AP with two TX antennas ,NIC 5300 with two RX  antennas , in other words I have a 2*2MIMO system.I have got the CSI matrix—2×2×30，but how to know the 1*1*30 CSI matrix represent of every  single transmit−receive antenna pair?thanks! @dhalperi @dpward

danha170, 2017-05-03:
Hi,
 First of all sorry I am asking irrelevant question in that thread.

But i want to know that the system you designed is fully MIMO-CSI ?

Is it possible that you can have 2 AP, two computers both transmit and receive at the same time, connected with each other, Getting CSI data as well. ?

lidongning, 2017-06-18:
you  said you  have a 22MIMO system,you have got the CSI matrix—2×2×30 , I have a 2×3 MIMO ,but I get csi includes 1×3×30 and 2×3×30,   how to set to get only 2×3×30? @linkzhao

danha170, 2017-06-19:
@linkzhao ...........Are you successfully getting CSI data fro 2 x 3 MIMO??
                          You did it on Ubuntu???
                          Have you made one PC a hotspot???
                         Could you give an idea how you establish the connection between two Pc's ?

danha170, 2017-06-19:
@lidongning ......

Are you successfully getting CSI data fro 2 x 3 MIMO??
You did it on Ubuntu???
Have you made one PC a hotspot???
Could you give an idea how you establish the connection between two Pc's ?

lidongning, 2017-06-19:
I use AP as AP , I  don't make hotspot  , but now I want to set it ..... my  NTX=2,NRX=3, I get 2*3*30 and 1*3*30 , but I want to only 2*3*30  @danha170

lidongning, 2017-06-19:
hostapd mode , need 2 pcs?    why hostapd connect router??? @danha170

danha170, 2017-06-19:
@lidongning ..........But if I use router It will give me output,,But I want to try 2 AP.......

I have a question , How you set two transmitters???
PC with NIC 5300, you can use two or three antennas as a RX..but how to set 2 transmitters???

lidongning, 2017-06-20:
sorry ,I only try a pc with 5300  and a AP @danha170

danha170, 2017-06-20:
@lidongning ............. Ok i got your point.....I want to talk to you in detail, could u send me your contact....e-mail...........whats app..anything? If possible

lidongning, 2017-06-24:
my email 1158907613@qq.com  @danha170   but I still don't get  Consistent data format 2*3*30 。