dhalperi/linux-80211n-csitool-supplementary #269: Why doe not work Mini PCI-E to USB Adapter (With SIM Card) Version 2.0 with Intel WiFi Link 5300 on it?

EuclidGH, 2016-12-12:
Hello anyone,

could somebody help me to run the adapter?
I want to use this adapter for mt Thesis, where I have to do some measurements with the help of USB adapter.
Neither Linux nor Windows recognizes the device. 
How can I run it? Did somebody have the same problem?

Thanking in advance.

Best regards,
Ieso

paulkirkland, 2016-12-15:
I have had the same problem, can't find a solution as yet. Though i am not sure how possible it would be unless you run it through USB 3.0 or greater...  If you ever find a solution post it up, i would be very greatful. 

In the mean time i am using either laptop with mini pcie, or a converter to pcie for pc both seem to work fine.

Paul

EuclidGH, 2017-01-20:
Hi, I couldn't find a solution and I have changed the USB to ExpressCard. adapter. It works great, but I use the Atheros CSI tool right now, because with Intel Card 5300 I had problems.

asifhanif1122, 2017-01-20:
@EuclidGH Can you please give details about how to extract CSI information from files collected using Atheros CSI tool? I installed Atheros tool but unable to seek guidance about CSI extraction from log files.

EuclidGH, 2017-01-20:
Hi, I just today got it worked. I can write the log data, but not extract it. I don't know how it works. There is also a directory "matlab", but I couldn't understand, how it works. 

I need also a solution for this step.

asifhanif1122, 2017-01-20:
@EuclidGH  I am also in pursuit of details about CSI extraction. 
This is my email address. "asif.hanif@outlook.com"
It would be nice thing if we share information about our mutual goal.

LifeRiver2017, 2017-11-27:
@EuclidGH I am also in pursuit of details about CSI extraction.
It would be a nice thing to share our information on this research topic.
you can give me your email address through emailing me: hitcsuzhb@gmail.com

Laurent-St, 2018-09-14:
@EuclidGH could you share the setup that you used and the way you got it working please? If you wish to communicate by email here is my address: laurent-st@hotmail.com
Thanks in advance!