dhalperi/linux-80211n-csitool-supplementary #272: can not setup injection or monitor mode

caichao, 2016-12-20:
I was trying to configure the network card as injection or monitor mode. When I run the setting script, it echo "command failed: Device or resource busy (-16)".  But when I run "iwconfig" or "ifconfig", I can see the mon0. Can somebody tell me why?

caichao, 2016-12-21:
It seems that run the command "iw mon0 set channel $1 $2" or "iw wlan0 set channel $1 $2" after the ifconfig or iwconfig will pop out no error messages. However, then the problem become "Error opening LORCON interface"

caichao, 2016-12-21:
It seems that all the problems have been solved. Here, I summarize the solutions to the problem that I have encountered.
(1) could not stop network interface.
command : "sudo /etc/init.d/networking stop"
result : stop: Job failed while stopping
solution: the command is no longer needed in the newer version of linux. You can shutdown the network use "ifconfig wlan0 down" instead.
(2)could not run the setup_inject.sh or setup_monitor.sh script
solution: run the following new script

for the receiver:
#!/usr/bin/sudo /bin/bash
echo "neccessary setup work"
sudo ifdown --exclude=lo -a && sudo ifup --exclude=lo -a
modprobe -r iwlwifi mac80211 cfg80211
modprobe iwlwifi connector_log=0x1
sleep 1
# Setup monitor mode, loop until it works
echo "trying to setup monitor mode"
iwconfig wlan0 mode monitor 2>/dev/null 1>/dev/null
while [ $? -ne 0 ]
do
	iwconfig wlan0 mode monitor 2>/dev/null 1>/dev/null
done

echo "open the network"
ifconfig wlan0 up

echo "set channel $1 & $2"
iw wlan0 set channel $1 $2

echo "successfully set the adapter"

for the transmitter:
#!/usr/bin/sudo /bin/bash
modprobe -r iwlwifi mac80211 cfg80211
modprobe iwlwifi debug=0x40000
ifconfig wlan0 2>/dev/null 1>/dev/null
while [ $? -ne 0 ]
do
	        ifconfig wlan0 2>/dev/null 1>/dev/null
done
iw dev wlan0 interface add mon0 type monitor
echo "open the adapter"
ifconfig mon0 up
sleep 1

iw mon0 set channel $1 $2
echo "successfully configure the adapter"

To check whether you have successfully configured it right, you can run iwconfig or ifconfig to see whether there is mon0 interface. It should have 5.32G band frequency and 15dB TX power. 

(3)"Error opening LORCON interface"
This is a very silly question. Just sudo run the random_packets, then it will be ok.

caichao, 2016-12-27:
More details can be seen at my blog: http://blog.csdn.net/caichao08/article/details/53894510

sunyu1122, 2017-01-03:
@caichao Do you set the hostap successfuly? I have set the hostap in a computer,but another computer cannot obtain the csi data.If you do ,i need your help

caichao, 2017-01-03:
@sunyu1122 Sorry, I haven't tried it.

sunyu1122, 2017-01-03:
@caichao Can you tell me why you setup injection or monitor mode?I read a paper and it also need to setup   
injection or monitor mode ,but i donnot know why it is.

caichao, 2017-01-03:
@sunyu1122 Injection or monitor mode enable multiple links simultaneously. This property is very appealing for indoor localization. And I have tested monitor mode and manage mode, the CSI signal in monitor mode is way much better than that in manage mode.

sunyu1122, 2017-01-03:
@caichao When you make experiment, can you need to realize the frequency-hopping

caichao, 2017-01-03:
@sunyu1122 I do not need to realize the frequency-hopping. I may know what you are doing. You may want to implement chronous, a single AP and TOF-based indoor localization method developed by MIT.

sunyu1122, 2017-01-03:
@caichao Yes,i do it .Can you tell me your research direction?I have  some questions to study for you.Can you tell me your email? Nice to meet you.Thanks

caichao, 2017-01-03:
@sunyu1122 Nice to meet you. My email address is caichao08@gmail.com.

sunyu1122, 2017-01-03:
@caichao ok,I write a message to your email and you can recieve it. Thanks

tahmidzbr, 2017-01-07:
Hi guys im also trying to do the chronos paper...can i join you? my email is tahmidzbr@gmail.com