dhalperi/linux-80211n-csitool-supplementary #275: package number is not right

luxiangx, 2016-12-31:
I set up monitor mode. And send package, but the received packages are less than the sent. Who can tell me what's the problem. Thanks.

asifhanif1122, 2017-01-01:
Are you using 2.4GHz band?

asifhanif1122, 2017-01-01:
I have also observed same phenomenon. But number of packets drooped are more in case of 2.4GHz channels as compared to channels of 5GHz band. One of the possible reasons is that 2.4GHz band is quite congested and many devices in our vicinity may be using this band. So due to interference , some of the transmitted packets get corrupted and dropped by receiver. On the other hand, 5GHz band is not so congested, in my case it was almost empty. I used to receive most of the transmitted packets with an average drop margin of 20 packets.

luxiangx, 2017-01-02:
@asifhanif1122 yes, I used the 5G band. However, the received packages were less than the sent. It can be the reason of using SOCK_DGRAM. I tried many times, and modified the code. But not helpful. I just want to synchronize time. Losing package will not guarantee the interval between packets index.

tahmidzbr, 2017-01-07:
@luxiangx I am not sure how you state packets are dropped, but in my case I am getting more packets all the time than i ping. Did you face something like this as well?

luxiangx, 2017-01-08:
@tahmidzbr In Ap mode, it's normal. You will receive some packages that are not recovered by the ping program.

tahmidzbr, 2017-01-08:
@luxiangx Thanks. Do you know if we can measure the CSI of  packets only transmitted from the AP? That is by sequencing or something?

luxiangx, 2017-01-08:
@tahmidzbr It's seem not possible unless you use monitor mode. Actually, you may also loss some packages in Ap mode if no other packages(not by ping) are received.

DontKnowMy, 2017-02-10:
I would like to know how you managed to get the packages numbers. Did you log them with the changed mode=0x5? Can you provide the needed changes in the matlab files?

luxiangx, 2017-02-17:
@DontKnowMy In monitor mode, the number of sending packages is set by yourself.

DontKnowMy, 2017-04-05:
I know but I want to distingish between the received packets. In my case CSI drops some packages too (around 20 in 10.000) compared to another script (just like tcpdump). Some packages of course can get lost but when the other script sees the packages they definitely arrive but the CSI log misses them!?!?