dhalperi/linux-80211n-csitool-supplementary #279: ping seem useless for the csi.dat

ZhaoZifei, 2017-01-10:
I have installed the tools ,finally when I run "sudo linux-80211n-csitool-supplementary/netlink/log_to_file csi.dat", I can get csi.dat and analyse it with matlab without ping. So I do not know what's wrong with my ping?

luxiangx, 2017-01-11:
@ZhaoZifei You can get more sample with ping. And this is the purpose of using ping.

chukku, 2018-05-28:
Sometimes I also face same situation. I can get the csi.dat without even pinging the router. Ok, Agreed that we get more samples with ping. But in my case I get more ping packets even if I ping a router which is unreachable. Getting more samples in this case doesnot make sense until it is coming from the intended router.