dhalperi/linux-80211n-csitool-supplementary #282: Nrx keeps changing (from 1 to 3)when I log csi data.

ZhaoZifei, 2017-01-16:
dear all:
         Nrx keeps changing (from 1 to 3,usually 1 ,sometimes 2 ,seldom 3)when I log csi data. But when I asked somebody else ,they says that Nrx is stable.So what's wrong with my job?Hoping someone to tell me how to set this Nrx. Thank you~~

ZhaoZifei, 2017-01-16:
I searched the forum and found the command " echo 0x4 | sudo tee /sys/kernel/debug/ieee80211/phy0/iwlwifi/debug/rx_chains_msk ", I don't know if that works and how to modify this command...

paulkirkland, 2017-01-19:
I have the same problems, the perm is set with the RSSI though even if i know which one should have the best signal strength the code sometimes disagrees... Just adapt your processing code to check the value of perm and assign is to the correct Rx

ZhaoZifei, 2017-01-19:
@paulkirkland 
Thank you for your answer!But what do you mean by THE VALUE OF PERM is that the 1×3 matrix for the transform of three antennas?If it is, how to processing it？
And I really want to know why the Nrx keeps changing(usually 1,sometimes 2,seldom 3...).By the way,my router is TP-LINK with only one antenna.

paulkirkland, 2017-01-19:
Yes, Perm is the matrix contained within the structure of the CSI. 
To process is just check the position one of the perm matrix perm(1) to see if values is either 1, 2 or 3 then assign this value to the correct Rx value in your extracted CSI matrix. 
It changes due to the RSSI signal strength at each antenna changing, with my set up i only have 2 recievers, and you can easily test how it works by covering one antenna, then noticing the perm values change order.... Although in testing with 2 antennas the value of perm can change (which is annoying). This is just due to the multipath of the wifi wave differing over time and as it reflects off of different surfaces.

DiDomenicoSimone, 2017-04-20:
Dear all,
I have the same problem as @ZhaoZifei. I could not understand why with a single antenna AP and the NIC 5300 the Nrx parameter changes dynamically for 1 to 3.

Any idea? Any suggestions?

Thanks a lot

saltdou, 2017-12-25:
@ZhaoZifei  I met the same problem as yours. Have you solved it?

DiDomenicoSimone, 2017-12-25:
@saltdou, after a deep analysis of the Intel driver, I found that the driver is not responsible for the changing of Nrx. After a long research I came out with the hypothesis that the low spatial diversity between the RX antennas changes the size of CSI matrix returned by the driver. For instance, if you are in a condition for which Nrx=1 and you try to increase the distance between the RX antennas, Nrx can switch from 1 to 2 or 3, depending on the distance.

saltdou, 2017-12-26:
Thanks for your reply. I have the same hypothesis, but don't confirm it so
far. I will keep you in touch if I solve this problem. Also, I found
another interesting scenario that shows the similar pi/2 fluctuation in
phase at 5GHz like 2.4 GHz but changes very occasionally. Did you meet
anything like this? I think the explicit beamforming exists across the
bands we haven't taken into account when extracting the CSI.