dhalperi/linux-80211n-csitool-supplementary #285: CSI Normalisation

DiDomenicoSimone, 2017-01-31:
Hi All,
I am currently working on CSI tool and CSI data, but I have a question about the CSI normalisation proposed in the Matlab script "get_sclaled_csi.m". 
Why does the CSI on each channel is normalised by the square root of its mean and multiplied by the square root of the total RSSI?.
In this way all CSIs have exactly the same power which is the total received power. 
Does it make sense from a physical point of view?.
I have also read the thesis of Daniel Halperin and says that after a trial-and-error test he found that normalisation. Anyway, I didn't understand how he verified that normalisation was right.  

Thanks a lot in advance for any explanations.

Simone

dhalperi, 2017-01-31:
The CSI matrix reported by Intel does not meet the standard. It is reported without taking into account amplification. However, most people expect the CSI to be in dBrn, and to get dB relative to noise you have to add in the RSSI, amplification, and noise.

The computations in there are just relatively logical expressions of the above:

* CSI matrix entries are in amplitude.
* amplitude squared is power ; mean is average power; square root of mean puts it back in amplitude units.
* RSSI - ADC - other constants gives the absolute units of the CSI matrix in dB.
* then add in noise floor.

Can you say more about "trial and error"? The main trained constant were in either the coefficients used for individual MCS or just picking `-92dBm` as a typical noise floor, but the formula itself is pure mathematical logic.

DiDomenicoSimone, 2017-01-31:
First of all, thank you for your fast reply.
Reading your reply I realised that I misinterpreted some line of code of the script "get_scaled_csi,m" and I understood the meaning of the normalisation you have done.
So, if I understood correctly, if the CSI matrix is divided by the square root of its power and multiplied by the square root of the RSS (total_RSSI - (AGC + Magic Number)) expressed in mW), the results will be the CSI matrix in amplitude unit taking into account the received power.
Is it right?

That said, what is the reference level of the CSI reported by the Intel (without any normalisation)?.
Referring to that reference level, what does the power on a CSI channel represent?   

Thanks a lot.

S.

p.s. : I read the "trial and error" in Chapter 5, Section 5.4.3, Note 1.

dhalperi, 2017-02-01:
p.s.: thanks, yep. It has been 5 years ;)

You can reverse the normalization procedure to get the meaning of the CSI reported by Intel.

Basically, the CSI report itself is measured on the AGC-controlled signal without actually taking into account the value of the AGC or the noise floor. And of course, there's a multi-level AGC before the DSP that generates the CSI, we have to take into account all those physical parameters to understand what the units really are.

DiDomenicoSimone, 2017-02-01:
Ok.
So after the normalization the CSI of each channel is expressed in dB relative to noise. 
In particular, if I just normalize each CSI channel dividing by the square root of the total CSI power (sum of power over the CSI matrix) and then multiplying by the square root of the total received power (RSS), the total power of the CSI matrix will be equal to RSS. 
RSS represents the total received power, hence it contains also the power contribution of the transmitted symbols.
That said, I would like to get the CSI going back through the receiver but without introducing the RSS.
At this end, in your opinion, would it be right normalize each CSI channel dividing (in linear scale) by the square root of the quantity (AGC*Magic Number Constant)?.
In this way, I should get the "real" power of each CSI channel, i.e. the power before the AGC.

Thank you in advance for your help.

S.

dhalperi, 2017-02-02:
I do not understand what units you want the final answer to be in.

DiDomenicoSimone, 2017-02-02:
I would like to get the CSI with same unit of CSI reported by Intel but without the effect of AGC.

dhalperi, 2017-02-02:
I don't think that's a meaningful concept -- because it can't be translated into any context. dBrn and dBm are two that make sense.

DiDomenicoSimone, 2017-02-02:
Generally, in an noiseless OFDM-based system, the channel frequency response is expressed in dB as it is the ratio between the DFT of the received symbol and the DFT of the transmitted symbol.
Normalizing the CSI matrix by the RSS or by RSS and noise, I obtain dBm and dBrn, respectively.
That said, I think that normalizing the CSI matrix by the AGC value and the magic number allows to get the CSI in dB scale.

dhalperi, 2017-02-03:
You cannot tell from a CSI matrix the power of the transmitted signal.

DiDomenicoSimone, 2017-02-05:
Yes, I perfectly know that. 

My reasoning was just for clarifying the unit of the CSI matrix and for telling that dividing the CSI matrix by the AGC gain I should obtain the CSI in absolute unit before the AGC.

Do you agree with me on that?