dhalperi/linux-80211n-csitool-supplementary #287: CSI Inject 2,4GHz doesn´t work but 5GHz works!?!?

DontKnowMy, 2017-02-06:
Hi,

I am having big trouble getting CSI work for the 2,4GHz channels. I can send and log CSI data on every 5GHz channel but none 2,4GHz channel works. All settings are the same but as soon as I just change to 2,4GHz the injection mode  doesn´t work. Packets doesn´t even leave the network interface.

Does anyone have a clue? I tried everything. Are there any known driver issues?

Thanks!

DontKnowMy, 2017-02-07:
Has no one experienced the same or is able to help? Would be very appreciated!!!

dhalperi, 2017-02-07:
Never seen this issue.

DontKnowMy, 2017-02-08:
so there shouldn´t be an error if I only change the channel with the iw command, right.

Workflow: setting up 5GHz with monitor interface, echo 0x1c113 for example to monitor_tx_rate, run randam_packets
-> changing to 2.4GHz with iw -> doesn´t work anymore

anybody an idea what could cause that problem? driver? card? 

Thanks

asifhanif1122, 2017-02-08:
Can you post snapshot of commands being used to set channel on 2.4GHz band.

DontKnowMy, 2017-02-08:
thanks for your interest in helping.

in the picture you see the sender on the left side. Setting everything up (running the scripts) first on 5GHz. Works! After that changing the channel. Packets don´t get sent. Same on the right side (receiver).

[image]

asifhanif1122, 2017-02-08:
I use following commands on transmitter and receiver respectively;
Transmitter:
[image]

Receiver
[image]

asifhanif1122, 2017-02-08:
You say , "**Packets do not even leave the network interface**".  How do you know that? If you send more than one thousand packets, you will see dots in printed message on terminal against each group of thousand packets. I have never seen problem with 2.4GHz band. Intel 5300 card supports 13 channels in 2.4GHz band and none of the CSI tool users ever mentioned the above stated issue.

DontKnowMy, 2017-02-09:
Hi,

thanks but I have tried all that before and I really know how to use the scripts but it has to be something different. 

I have used tcpdump to see whether the packets are leaving the interface or not. Capturing using tcpdump shows me the same information for 2.4GHz and 5GHz on the sender (injecting) but when I try to capture on another different device even with tcpdump no packages arrive with 2.4GHz. 

All channels in the 5GHz band work, only as soon as I choose a 2.4GHz channel it is not working. It drives me crazy.

dhalperi, 2017-02-09:
It looks to me like you're only trying very high rates (0x1c113) You should
test channel configuration out at an extremely reliable rate -- I seem
recall usually using 13 Mbps (is that 0x8101? It's been a while) as my
first experiment.

asifhanif1122, 2017-02-09:
Following table can help you change settings. It is in line with sir Halperi's instructions present on website.
[image]

DontKnowMy, 2017-02-10:
Hey thanks for the good advice. The low bitrates work also for 2.4GHz...:) Still mysterious why only MCS rate 0-2 work!?!? And even those rates don´t work reliable. When I am sending 1000 packets only around 400 arrive the client. 

Is there any other information stored somewhere which determines the rate like the iw dev set bitrates mcs-2.4 command? how is the iwlwifi driver configured?

dhalperi, 2017-02-10:
It could be antennas, antenna configuration, spatial diversity. Recall that
wavelengths are longer at 2.4 GHz so you need more spacing between antennas
to get MIMO performance.

If could also be that your antennas do not work well at 2.4GHz, but I doubt
it -- that usually goes the other way (2.4 but not 5GHz).

DontKnowMy, 2017-02-11:
I have build a setting with the right distances for the antennas. I still think it is something with software (driver).

[image]

DontKnowMy, 2017-03-29:
When I send packages on 2,4GHz with HT rates CSI gets collected but a simultaneous running tcpdump log shows me 11g. Should it not be 11n?

Thanks!

soltanaghaei, 2019-03-19:
Hi, I have a very similar problem and can't solve it in any ways. I wanna collect CSI values in 2.4Ghz with 40MHz bandwidth and a 3x3 setup (3tx antenna, 3 rx antenna). However, I don't receive any packets in most of the cases. I assume the problem is with the spatial diversity between the 3 antennas (Although I have ~6cm distance between antennas which is larger than the wavelength). Can you please guide me how you solved the problem in your cases and could you ever establish a robust communication with rates like 0x1c910-0x1c917?  @DontKnowMy, @asifhanif1122, @dhalperi, @dpward

TahsinaSanam, 2019-03-21:
@asifhanif1122  Can we use Router for transmitter and A laptop with 5300 NIC for receiver if we want to log CSI for 5GHZ?