dhalperi/linux-80211n-csitool-supplementary #295: why does the phase difference between the two antennas of Intel 5300 have a instantaneous change ?

sarahlsj, 2017-02-23:
[image]
In a silent environment, we think the CSI phase difference  between two antennas on a NIC should be stable  with time. But we find sometimes there is a sudden change in the phase difference like the above figure.  we wonder if you have ever seen this kind of phenomenon. And we will be very grateful if you can tell us the reason.

Korogodin, 2017-06-08:
@sarahlsj  Which mode did you use? Did you set a rate and a modulation for your transmiiter?

Korogodin, 2017-06-08:
@sarahlsj  Which mode did you use? Did you set a rate and a modulation for your transmitter?

sarahlsj, 2017-06-08:
we use the injection mode. 
And the packet rate of transmitter is 500 packet/s. 
About modulation, actually,  i  don't understand what you mean.  when we use csi-tool,we don't have anhy step to set a modulation for transmitter.

gesun, 2017-11-01:
there is a question that how to get the csi phase @sarahlsj