dhalperi/linux-80211n-csitool-supplementary #297: what are spatial multiplexing techniques being used in NIC 5300?

soltanaghaei, 2017-03-03:
Does anybody know what spatial multiplexing protocols are being used in NIC 5300? For example, when we set the flag_rate to use 3 antennas to transmit 2 or 3 streams, what kind of technique is used in this type of NIC? Do we have any control on that to change the used spatial multiplexing technique? 

@dpward @dhalperi 
I read [this paper](http://www.sigcomm.org/sites/default/files/ccr/papers/2010/January/1672308-1672313.pdf) that explains different spatial diversity or multiplexing techniques, but it's not clear to me how you test them (in commercial NICs?).

soltanaghaei, 2017-03-03:
@dhalperi I also studied the 802.11n protocol and it seems there are different frame formats for different diversity/multiplexing techniques, but I couldn't find out which one is used when :(

soltanaghaei, 2017-03-09:
@dhalperi, I've noticed that you're again active in this forum and I think you're the best person who can guide me in this regard. Did you have any chance to take a look at my question?

dhalperi, 2017-03-21:
Check out the MATLAB files with `_sm`. https://github.com/dhalperi/linux-80211n-csitool-supplementary/blob/master/matlab/get_scaled_csi_sm.m

closed 2017-03-21 by dhalperi

soltanaghaei, 2017-03-23:
@dhalperi  Thanks for your reply. It was very helpful, however, I couldn't figure out how the sm matrices are defined in sm_matrices.m even after reading IEEE 802.11n-2009. Can you please give me a clue to better understand how sm coefficients are chosen.

dhalperi, 2017-03-23:
Unfortunately I can't tell you the reasons why Intel chose those particular
matrices – you should research how spatial mapping matrices are chosen and
what properties they need to meet. However, it may help you to read chapter
4.5 of my thesis.

soltanaghaei, 2017-04-24:
Dear Daniel, I've noticed that the SM matrix in sm_matrices.m does not include the CSD shifting. Is there a way to reverse the effect of CSD shifting and measure the actual CSI matrix?

EsonJohn, 2018-07-12:
@dhalperi Is there any way that I can compensate the CSD shifting in the received CSI data? Is there any related code in the MATLAB files?