dhalperi/linux-80211n-csitool-supplementary #301: Set whitch mode to get CSI ?

HesperK, 2017-03-15:
Dear all,
I am a fresh man of CSI tools and I wanto collect CSI 20 packets per sec, I have a laptop(IWN 5300) as client and a TL-WDR 7500 router . Which mode should I set and how to set transmission rate (packets per sec)?
I will appreciate any responses. Thank you very much !
我有一个TL-WDR 7500 l路由器 和一个装有IWL 5300 网卡的电脑，想从电脑端以固定的速率收集CSI信号，我应该设置哪种模式？以及怎样以固定发包速率传输和手机CSI呢？跪谢各路大神！

linkin8834, 2017-04-02:
Hi,
       Just set a 802.11n network using your router (without authentication! Just use it as an open network), then, ping the router from your laptop with 5300 using 'ping -i x 10.0.0.1' where x is the desirable interval you want.  
       To get CSIs from iPhone, iPad, Macbook Pro, just set your laptop using hostapd (open network, no password), and download 'Ping Lite' on your mobile devices and ping your laptop. The CSI quality is pretty bad, though, except for MBP.

cbird7786, 2017-11-20:
Hi @linkin8834 , Could you please specify how the CSI data are got by ping Macbook to the laptop with 5300? Thanks in advance.

linkin8834, 2017-11-21:
Hi @cbird7786,
If you are using MBP, you can set your 5300 laptop as a software AP using hostapd and configured in the 5300 mode, i.e., with modified firmware and driver. Then, connect your MBP to the 5300 laptop AP. After that, you would have an IP address for the network on your MBP, say, 192.168.1.100. Finally, you can ping the 5300 AP by: ping 192.168.1.100. Your 5300 laptop shall report reception of CSIs when the MBP is pinging the 5300 laptop.

Maybe you can also specify the ping interval on MBP but I cannot remember. You can also try iperf.

cbird7786, 2017-11-22:
Hello @linkin8834 Thank you again for your valuable reply. My research domain is indoor localization. Maybe the injection model without any AP connection is a preferable option. I will try hostapd later. Hope to find some intersting hints.