dhalperi/linux-80211n-csitool-supplementary #302: Is it possible to collect CSI data on Desktop?

tfsanam, 2017-03-15:
Hi,

I am a PHD student and I need to collect CSI data for my research. I want to collect it on my desktop. Is it possiblee to Intel 5300 NIC on desktop? or there is a different device for desktop to collect CSI data? Also does the 802.11n CSI tool release work on desktop?

All I need for my research is a TP link router and a NIC equipped desktop to collect CSI data so that I can work directly on the CSI data.

I would really appreciate if you please guide me for this.

luxiangx, 2017-03-17:
Maybe you can try another CSI tool, Link: http://pdcc.ntu.edu.sg/wands/Atheros/
But I did not try it.

dhalperi, 2017-03-23:
Yes it works on Desktop just fine. As you can see from the papers we've written, as long as you can use the IWL5300 device the tool will work.

closed 2017-03-23 by dhalperi

tfsanam, 2017-04-07:
I have done upto step 5 as
"Once the driver is unloaded, reload it with CSI logging enabled:
sudo modprobe iwlwifi connector_log=0x1"

After that I got stuck on how to connect AP using IW utilities. I am novice in Linux.

Also upto the step that I proceeded, I noticed that my NIC card is not connected with the wifi network now. Is it ok? I mean before connecting with the AP with is as mentioned in the step 5, is it supposed to be not getting any wifi signal?

I would appreciate early response.

TahsinaSanam, 2019-03-20:
@dhalperi 

I have done upto step 5 as
"Once the driver is unloaded, reload it with CSI logging enabled:
sudo modprobe iwlwifi connector_log=0x1"

After that I got stuck on how to connect AP using IW utilities. I am novice in Linux. my Ap is is router mode with 802.11n only, encryption off. Still it says failure to connect.

Any help would be appreciated.