dhalperi/linux-80211n-csitool-supplementary #306: Hopping through the channels

guanchaoli, 2017-03-27:
Hi everyone, I am trying to du chronos with CSITOOL, but I am now stuck in some details.
1, The paper says it disables contention mechanism to enable fast channel switching, but I don't know how to do it.
2, The hopping time in paper is 2-3ms, but in my implementation, I use system('iw mon0 set channel xx ht20') to change the channel. It takes around 5ms. And extra time is needed to wait for the receiver to reply the packet with ACK information. So the actual period of my implementation is 10ms. Anyone knows how to reduce the hopping time? The chronos paper says it send the ACK directly from iwldriver, is that the same as using lorcon to inject? Do they wait for the socket?
3, I use lorcon(the tool used by the author of CSITOOL in injection mode) to send the packets, simply changing the first few bytes of payloads. But, the hopping protocol seems work badly because my NICs have difficulty receiving packets in some of the channels(mostly around 5.7GHz). In addition, the hopping is always reset in the first 13 channels due to timeout, which means the packets are not received in one of the channels in 2.4GHz. I send 10 packets every time to increase the success rate, but it seems useless. 
4, The phases of each antenna after unwrapping should be on a straight line, but in my hopping implementation, the phases rarely follow a line. The rate of getting a linear frequency-phase response is low. Is this normal? Is this related to the quality of antennas, the base plate of antenna, or the NIC itself? I found some of my NIC cannot receive a single byte in channel 32(in my area, this is the channel of highest frequency).

My descriptions may be complicated, but  the questions converge to a single one. Does anyone know how to implement a hopping protocol with low failure rate, and low phase shift?

Thank you for your reading. My email is guanchaoli@gmail.com or guanchaoli@icloud.com. If you are also doing similar works with chronos, we can discuss about the details.

@dhalperi @dpward

athpgf, 2018-07-08:
Hi!

I am also trying to implement Chronos, and have kind of the same problems as you. Did you manage to solve or find out why those results explained in 4? I have the same problem, most of the results are not linear.

In my case, in terms of packet loss, 2.4GHz performs way worse than 5GHz, I think is just because this band is really busy.

Thanks in advance!

seconrg, 2018-10-20:
Hello! I'm a undergraduate at USTC, sorry for my poor English.
I am also trying to implement Chronos, but I simply don't know how to change between channels like what the paper describes. Can you give me some advice?
Regards.