dhalperi/linux-80211n-csitool-supplementary #307: iwlwifi module compile error

veliogluh, 2017-03-28:
I am using Ubuntu 14.04.5 LTS, Trusty Tahr version without update. I am following the installation instructions. When I run the module compile command it ends with error. Below is the output. 

selcukpaker@mblack:~/linux-80211n-csitool$ make -C /lib/modules/$(uname -r)/build M=$(pwd)/drivers/net/wireless/iwlwifi modules
make: Entering directory `/usr/src/linux-headers-4.4.0-31-generic'
  CC [M]  /home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn.o
In file included from /home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn.c:51:0:
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn.h:567:15: error: ‘ieee80211_rssi_event’ defined as wrong kind of tag
          enum ieee80211_rssi_event rssi_event);
               ^
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn.h:567:15: warning: ‘enum ieee80211_rssi_event’ declared inside parameter list [enabled by default]
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn.h:567:15: warning: its scope is only this definition or declaration, which is probably not what you want [enabled by default]
make[1]: *** [/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn.o] Error 1
make: *** [_module_/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi] Error 2
make: Leaving directory `/usr/src/linux-headers-4.4.0-31-generic'
selcukpaker@mblack:~/linux-80211n-csitool$ 

I have also tried to compile with Ubuntu 15.4 and 16.4 version but I couldn't manage to it. I am following the instructions exactly. What am i missing?

happylily92, 2017-03-30:
In fact, at the beginning of the Installation Instructions, the are instructions as follows:

> These instructions are currently expected to work on Linux operating systems that are based on an upstream Linux kernel version between 3.2 (e.g. Ubuntu 12.04) and 4.2 (e.g. Ubuntu 14.04.4).

Therefore, maybe you should try to install CSI Tool in older version of Ubuntu.

veliogluh, 2017-04-05:
Hi,
I installed "12.04.5 LTS, Precise Pangolin" and tried again but I couldn't compile the source code. Below is the compile output.

selcukpaker@mblack:~/linux-80211n-csitool$ make -C /lib/modules/$(uname -r)/build M=$(pwd)/drivers/net/wireless/iwlwifi modules
make: Entering directory `/usr/src/linux-headers-3.13.0-32-generic'
  CC [M]  /home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn.o
  CC [M]  /home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.o
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c: In function ‘iwl_rs_rate_init’:
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c:2848:31: error: ‘struct ieee80211_conf’ has no member named ‘channel’
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c: At top level:
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c:3363:2: warning: initialization from incompatible pointer type [enabled by default]
/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c:3363:2: warning: (near initialization for ‘rs_ops.rate_init’) [enabled by default]
make[1]: *** [/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.o] Error 1
make: *** [_module_/home/selcukpaker/linux-80211n-csitool/drivers/net/wireless/iwlwifi] Error 2
make: Leaving directory `/usr/src/linux-headers-3.13.0-32-generic'
selcukpaker@mblack:~/linux-80211n-csitool$

Bowloforanges, 2020-01-22:
Hi there! hope this will help. I wasn't able to re-compile the driver until, out of desperation I tried doing it inside a virtul machine running ubuntu 14.04.1, It compiled no problem whatsoever. Maybe you can try the same.