dhalperi/linux-80211n-csitool-supplementary #312: my NIC card is not getting wifi link

tfsanam, 2017-04-07:
Hello,

I am using 5300 NIC on desktop and TP Link router TL-WR940N as my router. I have done upto step 5 as 
"Once the driver is unloaded, reload it with CSI logging enabled:
sudo modprobe iwlwifi connector_log=0x1"

After that I got stuck on how to connect AP using IW utilities. I am novice in Linux. So do notknow how to use iw utilities.

After the step 5 that I proceeded, I noticed that my NIC card is not connected with the wifi network now. Is it ok? I mean before connecting with the AP with iw utilities, as mentioned in the step 5, is it supposed to be not getting any wifi signal?

I would appreciate early response.

tfsanam, 2017-04-07:
I checked troubleshoot :

- this command from step 5 did not failed failed, : sudo modprobe iwlwifi connector_log=0x1

- I checked for the modified firmware. Look at the output of dmesg; I saw a line that contains the string: loaded firmware version 8.24.2.12

see below:

[68575.820557] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[68575.820566] iwlwifi 0000:03:00.0: Falling back to user helper
[68575.827055] iwlwifi 0000:03:00.0: Direct firmware load failed with error -2
[68575.827065] iwlwifi 0000:03:00.0: Falling back to user helper
**[68575.832040] iwlwifi 0000:03:00.0: loaded firmware version 8.24.2.12 op_mode iwldvm**
[68575.859571] iwlagn: connector callback registered
[68575.859627] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUG disabled
[68575.859632] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[68575.859636] iwlwifi 0000:03:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[68575.859642] iwlwifi 0000:03:00.0: Detected Intel(R) Ultimate N WiFi Link 5300 AGN, REV=0x24
[68575.861683] iwlwifi 0000:03:00.0: L1 Disabled; Enabling L0S
[68575.883477] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'

Please suggest mewhatcan I do now. Also my AP supports 802.11n, but I am not sure if its encryption mode is off. Can anyone give me any idea how can I check that my AP is delivering 802.11n?

thanks

closed 2017-04-11 by tfsanam

adlinaadha97, 2020-02-24:
hi, how did u solve this?