dhalperi/linux-80211n-csitool-supplementary #317: Difficulties installing CSI tool

mirasarkis, 2017-04-18:
Hello,
I am trying to follow the installation steps following the site (https://dhalperi.github.io/linux-80211n-csitool/installation.html).

I am stuck in Step 2, where I have to merge my modifications into Linux ('Tip' section).
I am running the Ubuntu 4.2.0-42.49 kernel image to which I cannot associate a correspondent UBUNTU_KERNEL_TAG from http://people.canonical.com/~kernel/info/kernel-version-map.html .

The error I am getting is "Fatal: couldn't find remote ref Ubuntu-4.2.0-42.49 .

One solution would be changing my kernel image and replacing it with another from the above site. 
I would like to make sure that there is no other solution for this.

Thank you in advance,
Mira

AndroidUnited, 2017-04-19:
Hello Mira and everyone else

I am also trying to install the CSI tool on my laptop but facing the exact same issue as mentioned above. I have installed Ubuntu 14.04.5 with kernel 4.4. I am sure this is unsupported as the website says upto v4.2. As a result I installed Ubuntu 12.04.5 with kernel 3.13.x.x, which is unable to "git pull --no-edit" for some reason. Which version of ubuntu/kernel image should i install in order to make this tool work?

@dpward sir, or anyone else, if possible please try to answer this question about installation. Thanks you.

mirasarkis, 2017-04-19:
Hello Akarsh,

In my case, my kernel tag does not correspond to this:
UBUNTU_KERNEL_TAG=Ubuntu-x.y.z-x.y.

Other tag formats  are defined in the CSItool.
You can check these tags by running the following command:
                >>git fetch ubuntu --tags
The output would be an exhaustive list of available list corresponding to
ubuntu kernels.
To refine your search,  you run for example:
               >>git tag -l Ubuntu-lts-4.*
Note: in my case I  have ubuntu 14.04.5 LTS with a kernel version 4.2.
The output of this second command is a list of tags; at least one should
correspond to your version.
This tag becomes the value of ${UBUNTU_KERNEL_TAG}

Continue with: >>git pull --no-edit ubuntu ${UBUNTU_KERNEL_TAG}.
And it's done!
I hope this might help you.

Note: I saw that you downgraded your ubuntu version to 12.04.5. In fact, It
was possible to keep the newer version and install a 4.2 core image and
headers using this command:

         >>sudo apt-get install linux-generic-lts-wily

And the remove the 4.4 core image and header by first identifying the list
of installed images and headers

        >>dpkg -l | grep linux-image

        >>dpkg -l | grep linux-headers

To delete image/header:

        >> sudo apt-get purge linux-image-xx linux-header-xx

Regards,

Mira

AndroidUnited, 2017-04-19:
Thanks for the reply Mira. Right now I have followed your advice and installed 14.04.1 with the kernel version - Linux student 4.2.0-42-generic #49~14.04.1-Ubuntu

So i am trying the "git fetch ubuntu --tags" but git throws the following error -

fatal: 'ubuntu' does not appear to be a git repository
fatal: Could not read from remote repository.
Please make sure you have the correct access rights
and the repository exists.

Can you please check if there is a typo in your answer or point out the mistake that I am doing.

Thank you very much,
Akarsh

mirasarkis, 2017-04-20:
Make you sure first that you are inside the linux-80211n-csitool directory, where your .git repository resides.
Then try to fetch tags using the command "git fetch ubuntu --tags". (I confirm there is no typo in this command).

Note: To build the modified driver, make sure you are also in linux-80211n-csitool directory.

Regards,
Mira

AndroidUnited, 2017-04-29:
Hello guys

Thanks to mira, I have successfully installed the tool, but having different issue (which seems to be a very common issue and the faq is not helpful in solving it)

1. I have installed the linux csi tool on my laptop which is running 14.04.01 LTS with 3.13.32.57 generic linux kernel.
2. I have also disabled by my AP encryption, its in 802.11 n mode only.
3. I am able to ping successfully on both the router and the laptop from each other.
4. I also have set the rate to the HT rate (54 mbps).
5. Ping is 100% success with zero packet loss.

**However the csi file is empty and is unable to log anything**.. i have followed all the instructions to the point and been searching for solution for over a week.

I would really appreciate any help in this regards.

Thank you,

mirasarkis, 2017-05-04:
Hello 
I am also having the same problem for collection CSI between my Client and my HostAPd.
I have some doubts that maybe I can share:

- On the client: the content of /sys/../macAP/rate_scale_tabel, shows an lq Type HT SISO 40MHz 
But  On the AP: the content of /sys/../macClient/rate_scale_tabel shows lq type legacy. 
I am not sure if these asymmetric  channels are the reason why I cannot log CSI.

- When I try to fix the transmission rate on either or both my Client and my AP, my AP freezes and the connection is cut.  I don't really understand why?!

- An "80211n station" is mentioned  here: [https://dhalperi.github.io/linux-80211n-csitool/installation.html](url).  I know the Intel 5300 is capable of doing 80211n, but it can also make 80211 a, b and g. How can I make sure that my client is working in 80211n mode?
Is it enough to add the Greenfield mode to the hostapd configuration, to restrict the AP connections to only 80211n stations?

Any help please?
Thank you in advance

AndroidUnited, 2017-05-04:
Hello Mira

1. I was able to fix my issues and successfully log csi data. However, I am using a TPLINK router with openwrt, to connect my laptop. Its very easy to change protocol this way.
2. I would also recommend you to use a router for this purpose to avoid unnecessary complications. You can simply use your home router (unplug internet wire).
3. I set my router in 5Ghz HT mode which is 802.11 n and disabled encryption.
4. Set rate any rate higher than 6mbps. I used 54mbps. (feel free to convert that to hexadecimal).

Hopefully, this works for you.

Regards,

mirasarkis, 2017-05-05:
This means that the card of my access point should not be obligatory an
Intel 5300 in order to collect CSI data??

Is it enough that only my receiver has the Intel wifi card deployed with
the modified iwlwifi driver?

Mira

AndroidUnited, 2017-05-08:
Yes, only the receiver should be a laptop with Intel 5300 chipset to collect the CSI. You can use any router as a transmitter.

tfsanam, 2017-05-18:
@AndroidUnited 

How did you router in 5Ghz HT mode ?
what are the associated changes in router setting and NIC card setting?

Thanks.

mirasarkis, 2017-05-18:
How are you setting up your router?

Are you using hostapd?
In hostapd, it is simple..you just need to set the channel to 36 or above!
But first, you should verify that your Intel 5300 can send beacons on the
5Ghz frequencies. If you find "passive scanning no ibss" or "no ir" or
"disabled" flags when you run "$iw list" command, then it is not possible
to make your card work in AP mode on 5Ghz.

Hope this helps!
Mira

zhangdh24, 2017-07-03:
@AndroidUnited  Hi, I have the same problem as you mentioned above. I install csitool running 14.04.01 LTS with 3.13.32.57 generic linux kernel. I can ping the router successfully but the csi file is empty. Can you tell me how do you fix the problem? Thanks a lot.

chukku, 2018-01-08:
Hello, 

I am also trying to install CSI tool as mentioned in the installation instruction. But in step 2 when i give the command 👍 CSITOOL_KERNEL_TAG=csitool-$(uname -r | cut -d . -f 1-2)
i receive an error saying that csitool is not a valid command. I have followed all the steps mentioned. But I don't know how to resolve this issue? 

Any inputs would be highly appreciated. 
Thanks,