dhalperi/linux-80211n-csitool-supplementary #319: CSI for 1x3 SIMO link

DiDomenicoSimone, 2017-04-20:
Hello everyone,
I am using CSI tool to collect CSI data from the Intel NIC 5300.
My setup consists of a single antenna (single stream) Access Point and an Intel NIC 5300 equipped with 3 antennas. Collecting the data I noticed that some captured packets have a CSI matrix of size 1x3x30, others have 1x2x30 and others 1x1x30. This happens only using a single antenna AP or a multiple antenna AP forced to work with a single spatial stream.

Any ideas? Any suggestions?.

Thanks a lot in advance.
Simone

DiDomenicoSimone, 2017-05-26:
Hi @DneprovV,
during my experiments I manually disabled all the settings that can change the number of RX antennas, like the power saving feature. By doing a campaign of experiments I noticed that with the same Access Point and the same receiver the number of RX antennas can change with the tested environment. There can be 3 possible kind of reasons for disabling some of the receive antennas:

- Crosstalk between antennas due to the imperfect design of the printed circuit. 
- High RSSI imbalance between antennas;
- Not enough spatial diversity between antennas;

I excluded the crosstalk because in that case I should have ever had the same number of antennas  regardless of the environment. 

I excluded the RSSI imbalance because I put some debug prints in to the iwlwifi driver and chacked that the RSSI difference between the antennas was not greater than the threshold (50 dB, If I remember correctly). 

At last, I think with reasonable probability that the variability in the of receive antennas is due to the spatial diversity and therefore due to the frequency selectivity of the sensed environment and the distance between antennas. You can experience this effect if you try to increase the distance between antennas or move to a larger environment (hopefully more selective).

Simone

ozawali, 2017-07-20:
you should configure the tr antenna and sptail steram both.

junsujang, 2017-08-04:
In #57 some people mention kernel version as one of the possible reasons for such behavior. Have you looked into this possibility as well?

DneprovV, 2017-08-04:
Hello, @junsujang. Thanks for pointing on that problem, but it seems it's not my case. I'm using kubuntu-14.04.3-desktop-amd64. Can't tell you kernel version, I'm away from my work computer. But anyways, I desided to use 2 Intel WiFi 5300 as transmitter and receiver with injection mode. As far as I can say (I'm not very experienced in wireless communications), it works well. I get CSI matrix with size I expect to get.

junsujang, 2017-08-04:
Hi, @DneprovV 
Thanks for getting back to me.
So do you have any idea as to how your original issue of varying number of antennas got resolved? You mention that you have decided to use injection mode for both Tx and Rx; do you mean calling setup_injection.sh (Tx) and setup_monitor_csi.sh (Rx)? How did you set things up initially?

I really appreciate your reply comments.

DneprovV, 2017-08-05:
Well, at my fisrt message of this thread my setup was like this: commercial AP as TX and Intel 5300 as RX. So, my hypothesis is that I couldn't manage TX rate and commercial AP choose rate and modulation scheme on it's own. That's why I decided using 2 Intel 5300. So I just made instructions in README file in injection folder of *supplementary project. Also original scripts setup_injection.sh (Tx) and setup_monitor_csi.sh (Rx) didnt work on my computer well, so I found some modification here http://m.blog.csdn.net/u012840636/article/details/53894510. There is just some sleep 1 addition to original scripts, but it works for me.