dhalperi/linux-80211n-csitool-supplementary #323: Collecting CSI data on 5GHZ - Intel5300 does not allow it!

mirasarkis, 2017-05-17:
Hello,
I am trying to setup hostapd access point to work on the 5GHz frequencies, but the Intel 5300 wifi card does not allow me!
My final objective is to collect CSI data on the 5GHz channels.

$iw list : shows that the 5GHz channels are either disabled, or flagged as "passive scanning, no IBSS".

 Is there someone that had a similar problem? If yes, what could be a solution (if there is one!)?

Thank you in advance for any help,

Regards,

Mira

ozawali, 2017-07-20:
Maybe the configuration are not right in the hostap

TahsinaSanam, 2019-03-21:
@mirasarkis 

Did you solve the problem? would you please help me indetail how to get 5GHz CSI data?