dhalperi/linux-80211n-csitool-supplementary #331: why can't get 2*3*30 csi?

lidongning, 2017-06-14:
@dpward @dhalperi my AP Ntx=2, my pc  Nrx=3, but I get CSI includes 1*3*30 and 2*3*30. I ping 2500/s in my pc.  I wnt to get   2*3*30 CSI , how can I do ?

dhalperi, 2017-06-15:
The dimensionality is (tx streams) x (rx antennas) x (30 subcarries). So when you get 1 x 3 x 30 CSI, that's because the transmitter used 1 stream, aka MCS 0-7.

dhalperi, 2017-06-15:
Follow the FAQ to force 2-stream rates.

closed 2017-06-15 by dhalperi

lidongning, 2017-06-16:
Tanks~~  when I  do :
sudo cat /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/00:0f:34:5d:01:ff/rate_scale_table
soemtime it is lq type legacy, sometme  it  is lq type HT?   but only in HT ,it provides CSI, how can I do?

lidongning, 2017-06-16:
@dhalperi Tanks~~ when I do :
sudo cat /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/00:0f:34:5d:01:ff/rate_scale_table
soemtime it is lq type legacy, sometme it is lq type HT? but only in HT ,it provides CSI, how can I do?

lidongning, 2017-06-16:
I set  0xc10B rate, it is 2-stream,  but CSI still have 1-stream.  what happen?  @dhalperi

dhalperi, 2017-06-16:
I do not think you have set the rate correctly if this is happening.

lidongning, 2017-06-17:
I set rate 0xc90e, 0xc90c,  0x1c90e,  I can't get only 2-stream,  always include 1-stream.   my AP also set 80211n mode , I don't understand 。。。。so depressed。。。。 @dhalperi

ozawali, 2017-07-18:
@dhalperi  you are right, and I configure successfully according to the FAQ, but I have another question about the CSI maxtrix that is 3x3x30, So what's the antenna pairs respective to the measurement matrix. Thank you very much!

lidongning, 2017-07-18:
my Ntx=3, Nrx=3, could you help me to set correct rate  in order to get 3x3x30 ???   I have try many rates ..... @ozawali

ozawali, 2017-07-20:
if you set correctly, the receiver can get the matrix is 3x3x30. As for the rate, you can set according to the MCS 0-23.

evaszx, 2019-07-25:
@dhalperi Hi ,I am being in the problem of the size of csi. I have a router with 3 antennas,but I get the size of data is 2x3x30, then I set the transmit rate and antennas to 0x420A according the FAQ#1, but I still get the 2x3x30 size of csi data. I don't konw why the phomenon is so confusing. Could you help me ? Thank you!
The content of the "rate_scale_table"is with the real-time change of 16 possible transimit rates:
sta_id 0
failed=0 success=0 rate=0FFF
fixed rate 0x0
valid_tx_ant ANT_A,ANT_B,ANT_C
lq type HT
 MIMO2 40MHz   
last tx rate=0x4001
general:flags=0x0 mimo-d=5 s-ant0x2 d-ant=0x6
agg:time_limit=4000 dist_start_th=3 frame_cnt_limit=63
Start idx [0]=0x0 [1]=0x0 [2]=0x0 [3]=0x0
 rate[0] 0x1890F 60mbps (64QAM 5/6)
 rate[1] 0x1890F 60mbps (64QAM 5/6)
 rate[2] 0x1890E 54mbps (64QAM 3/4)
 rate[3] 0x1890E 54mbps (64QAM 3/4)
 rate[4] 0x1890E 54mbps (64QAM 3/4)
 rate[5] 0x4001 48mbps (64QAM 2/3)
 rate[6] 0x800B 36mbps (16QAM 3/4)
 rate[7] 0x10009 24mbps (16QAM 1/2)
 rate[8] 0x4007 18mbps (QPSK 3/4)
 rate[9] 0x8005 12mbps (QPSK 1/2)
 rate[10] 0x1026E 11mbps (QPSK CCK)
 rate[11] 0x400F 9mbps (BPSK 1/2)
 rate[12] 0x800D 6mbps (BPSK 1/2)
 rate[13] 0x10237 5.5mbps (BPSK CCK)
 rate[14] 0x4214 2mbps (QPSK DSSS)
 rate[15] 0x820A 1mbps (BPSK DSSS)

lidongning, 2019-07-25:
sorry.My problem hasn't been solved.------------------ 原始邮件 ------------------

 
&mdash;