dhalperi/linux-80211n-csitool-supplementary #338: Install - Connection Issue

Kolson3208, 2017-06-28:
I have installed the software on an Ubuntu 3.13.0-32.57 version and seem to be getting stuck in step 5.
Currently, I am using the Network Manager for WiFi management.
All parts 1-4 seem to install without issue.  Step 5 - I am able to unload iwlwifi driver.. ( All wifi signals go away.. as expected and connection work before unloading)  I am able to reload driver with logging ( all wifi signals show up again) but am unable to connect after drivers are reloaded.   

*WiFi networks have no encryption. 

Please advise. 
Is there a way to determine if steps 1-4 were really successful? WiFi connection log errors? etc?

ZhengChang467, 2017-06-30:
I have the same question!!

dhalperi, 2017-06-30:
Please read [FAQ#7](http://dhalperi.github.io/linux-80211n-csitool/faq.html).

dhalperi, 2017-06-30:
Per the instructions, you should uninstall/disable network manager and instead configure wireless state manually.

ozawali, 2017-07-21:
please read the faq!

adnaneGdihi, 2017-07-31:
Hi @Kolson3208 @ZhengChang467  ,
have you checked yourAP? in fact the modified iwlwifi can **only** connect to **802.11n** AP. mixed modes like 802.11a/g/n doesn't work so, try to configure your AP to work on 802.11n. 
i hope this works for you

doudou123456, 2017-09-15:
I have the same question!  Do you solve it?
thank you！！

rt6n1, 2018-04-15:
I have exactly the same problem:
- using Ubuntu 14.04 LTS, 3.13.0 kernel
- AP: 802.11n only, no security
- laptop initially connects to the AP
- unload the driver, connection lost (as expected)
- reload the driver with CSI logging enabled (connector_log = 0x1), cannot reconnect
Please advise!