dhalperi/linux-80211n-csitool-supplementary #339: I can't change rate according to FAQ??

lidongning, 2017-07-07:
echo 0x1C117 | sudo tee >     /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/dc:fe:18:64:fc:94/rate_scale_table

but it shows :  bash: /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/dc:fe:18:64:fc:94/rate_scale_table: Permission denied   ??? @dpward @dhalperi 

I do :  sudo cat /sys/kernel/debug/ieee80211/phy0/netdev:wlan0/stations/dc:fe:18:64:fc:94/rate_scale_table
  ta_id 0
failed=0 success=0 rate=0FF0
fixed rate 0x0
valid_tx_ant ANT_A,ANT_B,ANT_C
lq type HT
 MIMO3 20MHz   AGG on
last tx rate=0x1C116
general:flags=0x0 mimo-d=5 s-ant0x4 d-ant=0x3
agg:time_limit=4000 dist_start_th=3 frame_cnt_limit=63
Start idx [0]=0x0 [1]=0x0 [2]=0x0 [3]=0x0
 rate[0] 0x1C117 60mbps (64QAM 5/6)
 rate[1] 0x1C117 60mbps (64QAM 5/6)
 rate[2] 0x1C116 54mbps (64QAM 3/4)
 rate[3] 0x1C116 54mbps (64QAM 3/4)
 rate[4] 0x1C116 54mbps (64QAM 3/4)
 rate[5] 0x4001 48mbps (64QAM 2/3)
 rate[6] 0x800B 36mbps (16QAM 3/4)
 rate[7] 0x10009 24mbps (16QAM 1/2)
 rate[8] 0x4007 18mbps (QPSK 3/4)
 rate[9] 0x8005 12mbps (QPSK 1/2)
 rate[10] 0x1000F 9mbps (BPSK 1/2)
 rate[11] 0x400D 6mbps (BPSK 1/2)
 rate[12] 0x800D 6mbps (BPSK 1/2)
 rate[13] 0x1000D 6mbps (BPSK 1/2)
 rate[14] 0x400D 6mbps (BPSK 1/2)
 rate[15] 0x800D 6mbps (BPSK 1/2)

closed 2017-07-17 by lidongning

evaszx, 2019-07-25:
@lidongning I have the same problem with you, have you solved the problem, could you point me out of  the problem? thank you very much!

lidongning, 2019-07-25:
sorry .My problem hasn't been solved. ------------------ 原始邮件 ------------------

 
&mdash;