dhalperi/linux-80211n-csitool-supplementary #341: can I use Desktop PC with intel 5300  instead of laptop??

lidongning, 2017-07-18:
when I use Desktop PC with intel 5300 and TL_WDR7500,  it speed only 6M??

ozawali, 2017-07-20:
denpend on what the model AP used, so if you set the 5300 as the client and the TL_WDR as the AP the speed related to the current state. if you want control the rate I suggest you use the injection mode.

ozawali, 2017-07-20:
please read the faq carefully[image]

lidongning, 2017-07-20:
Tank you very much !

nafiul-nipu, 2017-07-20:
hi.. i am trying to set up injection mode.. but when i run the scripts it says permission denied for both transmitter and receiver.. can you suggest me regarding this problem??

ozawali, 2017-07-21:
please read the faq carefully, and then you will reach it.

nafiul-nipu, 2017-07-29:
@ozawali  hello.. i have another problem..  for getting csi data.. in the example it shows that the csi data is a matrix of 1x3x30   but i am getting 2x3x1652 , 2x3x1456 etc. means i am getting a lot of data.can you please tell me how i would get 1x3x30. i am using a desktop and intel 5300 nic inside it and a TPlink router with two antennas.

scivision, 2017-11-05:
Link to the FAQ above got mangled:

https://dhalperi.github.io/linux-80211n-csitool/faq.html