dhalperi/linux-80211n-csitool-supplementary #348: Received CSI is in the form of 2x3x30 matrix, what is 2 here?

sapansanu, 2017-09-29:
I am receiving the CSI in the form of 2x3x30 matrix where 30 is the number of subcarriers and 3 is the number of antennas.  In the documentation, the data is in the form 1x3x30. Why am I getting a 2x3x30 matrix? And what is 2 here?

sapansanu, 2017-09-29:
Also, for processing the CSI, data from which of the 3 antennas should be used? Or should I aggregate data form all antennas together?

asifhanif1122, 2017-09-30:
2x3x30 shows that number of Tx antennas is 2. What are you using as transmitter? Laptop or router?
In case of laptop or desktop computer as transmitter, you must have been using Intel 5300 card. 2x3x30 may have arisen due to number of Tx antenna selection bits in settings. If router is being used, it may be adapting its transmission rate based on MiMo and hence number of Tx antennas gets changed accordingly.

gesun, 2017-11-02:
I want to know where the CSI comes from when I want to recieve two kind of CSI messages? I want to know the IP of router that can send the   @asifhanif1122 @dhalperi @dpward @ippeb

Challenger1132, 2017-11-28:
@asifhanif1122 
In the transmitter and receiver, they are Intel 5300 NIC and both have 3 antennas working in monitor mode.
Inspecting the CSI matrix, they are always 1*3*30. How can I obtain CSI data by 3*3*30 ?

hminkoi, 2019-04-17:
> @asifhanif1122
> In the transmitter and receiver, they are Intel 5300 NIC and both have 3 antennas working in monitor mode.
> Inspecting the CSI matrix, they are always 1_3_30. How can I obtain CSI data by 3_3_30 ?

Actually, the 1x3x30 and 3x3x30 is because of your set-up... You have only 1 transmitter antenna. You can see this more on FAQ 2. Thus, you need 3 antennas transmitter to get 3x3x30. @Challenger1132

evaszx, 2019-07-25:
@hminkoi Hi, I am being in the problem of the size of csi. I have a router with 3 antennas,but I get the size of data is 2*3*30, then I set the transmit rate and antennas to 0x420A according the FAQ#1, but I still get the  2*3*30 size of csi data. I don't konw why the phomenon is so confusing. could you help me ? thank you!

evaszx, 2019-07-25:
I am being in the problem of the size of csi. I have a router with 3 antennas,but I get the size of data is 2x3x30, then I set the transmit rate and antennas to 0x420A according the FAQ#1, but I still get the  2x3x30 size of csi data. I don't konw why the phomenon is so confusing.

evaszx, 2019-07-25:
@hminkoi  Hi ,I am being in the problem of the size of csi. I have a router with 3 antennas,but I get the size of data is 2x3x30, then I set the transmit rate and antennas to 0x420A according the FAQ#1, but I still get the 2x3x30 size of csi data. I don't konw why the phomenon is so confusing.

evaszx, 2019-07-25:
@hminkoi Hi ,I am being in the problem of the size of csi. I have a router with 3 antennas,but I get the size of data is 2x3x30, then I set the transmit rate and antennas to 0x420A according the FAQ#1, but I still get the 2x3x30 size of csi data. I don't konw why the phomenon is so confusing. Could you help me ? Thank you!

soltanaghaei, 2019-11-16:
@evaszx I have a similar problem, could you figure out what the issue was?