dhalperi/linux-80211n-csitool-supplementary #357: iwlwifi module compile error

hanspond, 2017-12-20:
Version: Ubuntu 14.04.1 LTS
version_signature Ubuntu 3.13.0-32.57-generic 3.13.11.4

the following error appeared when I tried to compiled the driver. What could be wrong about this?
Previously tried on a Ubuntu 14.04.04 but got the same error.

'''
make: Entering directory `/usr/src/linux-headers-3.13.0-32-generic'
  CC [M]  /home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.o
/home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c: In function ‘iwl_rs_rate_init’:
/home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c:2848:31: error: ‘struct ieee80211_conf’ has no member named ‘channel’
  sband = hw->wiphy->bands[conf->channel->band];
                               ^
/home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c: At top level:
/home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c:3363:2: warning: initialization from incompatible pointer type [enabled by default]
  .rate_init = rs_rate_init_stub,
  ^
/home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.c:3363:2: warning: (near initialization for ‘rs_ops.rate_init’) [enabled by default]
make[1]: *** [/home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi/iwl-agn-rs.o] Error 1
make: *** [_module_/home/hp/linux-80211n-csitool/drivers/net/wireless/iwlwifi] Error 2
make: Leaving directory `/usr/src/linux-headers-3.13.0-32-generic'
'''

Bowloforanges, 2020-01-22:
Hi there! hope this will help. I wasn't able to re-compile the driver until, out of desperation I tried doing it inside a virtul machine running ubuntu 14.04.1, It compiled no problem whatsoever. Maybe you can try the same.

iftolito, 2020-02-01:
> Hi there! hope this will help. I wasn't able to re-compile the driver until, out of desperation I tried doing it inside a virtul machine running ubuntu 14.04.1, It compiled no problem whatsoever. Maybe you can try the same.

Please can you send a screenshot of you instalation instruction because i try it in 14.04.1 and i have also the same problem and my file CSI.DAT is always empty. thank you

Bowloforanges, 2020-02-05:
Of course. Keep in mind I used the OS and kernel specified on the last response, but inside a virtual machine just to compile it, then proceeded with the later steps in my 'physical' pc.

Here are my steps, pretty much the same as the ones on the installation guide.

[commands-14.txt](https://github.com/dhalperi/linux-80211n-csitool-supplementary/files/4157035/commands-14.txt)

PS: I have not been able to log csi data yet (not that I tried too hard but when I get it working I'll let you know!)