dhalperi/linux-80211n-csitool-supplementary #360: csi.dat is empty(log_to_file is not working)

bear1704, 2018-01-11:
I checked all the previous question list,and it did not really give me a solution to my trouble.

Installing the firmware, I follwed the instruction guide(including those tips)  
However at the last step, the csi.dat was empty.

Assuming log_to_file.c is not catching the replying data from AP.   
 

[https://github.com/bear1704/etc/blob/master/file1.jpg?raw=true](url)

[https://github.com/bear1704/etc/blob/master/file2.jpg?raw=true](url)

[https://github.com/bear1704/etc/blob/master/file3.jpg?raw=true](url)

[https://github.com/bear1704/etc/blob/master/file4.jpg?raw=true](url)

samuel920119920119samuel, 2018-05-01:
@youngle37 and I also encountered this problem and we finally solved it.
Check #362 out! Hope it will solve your problem!

764398256, 2018-11-08:
How do you solve the problem ?
I also encountered this problem in AP mode .