dhalperi/linux-80211n-csitool-supplementary #362: CSI log file is empty

michaelp123, 2018-02-01:
Hi Guys, 
I have successfully installed the tool, but having different issue:

I am able to ping successfully on both the router (hostapd @ 2.4 ghz, intel5300) and the client (samsung device) from each other.
I also have set the rate to the HT rate (60 mbps).
Ping is 100% success with zero packet loss.
However the CSI file is empty and is unable to log anything.
I couldn't find any solution that worked for me in this Q&A.

I would really appreciate any help in this regards.

Thank you,

rt6n1, 2018-04-15:
I'm experiencing exactly the same problem:
- Installation went OK
- Connects to 802.11n no security AP (ZTE ZXHN H168N V3.1), HT rate, 40 MHz BW
- Pings AP
- No messages in the CSI terminal, CSI file empty.
Can anyone please advise?

youngle37, 2018-04-30:
I'm using ubuntu 14.04 with kernel version 3.13

I solve this problem by adding this in AP's hostap configuration file
```
# hostapd.conf
ieee80211n=1
```

but before this, you have to type this
```
dmesg | grep iwlwifi
```
to check if the output with a line that contains the string "loaded firmware version 8.24.2.12"
If not, that's mean the modified firmware isn't loaded.

rt6n1, 2018-05-12:
Thanks, @youngle37! I reverted to Ubuntu 12.04 and everything worked right from the start.

ljk8918, 2018-06-14:
Hi @rt6n1 
I've met the same problem.
You said you revert to 12.04, still have you worked fine?
regards

rt6n1, 2018-06-14:
Hi @ljk8918,

CSI logging works fine. The only trouble is that data cannot be processed in GNU Octave that old, so I have to transfer the files to a PC running Ubuntu 16. This is indeed a nuisance.

zhouyu9712, 2018-09-15:
I have the same
problem with you.  need  help.

rt6n1, 2018-09-15:
It simply doesn't work on Ubuntu 14. I reverted to 12 and data logging
worked fine. For processing I had to switch to a newer Ubuntu.

Regards,
Milan