dhalperi/linux-80211n-csitool-supplementary #364: CSI not logging

struggletown, 2018-02-06:
Sorry to start a new thread on this, but I am also struggling with the loading of CSI data to file. When running the final step in the installation guide, log_to_file runs, but doesn't load data. 

From browsing these threads and the FAQ, I have found that:
- transmission is occuring at HT as required (using instructions in FAQ 1 to get the rate_scale_table)
- the firmware is correctly loaded ('dmesg | grep firmware' returns 'loaded firmware version 8.24.2.12')
- the laptop/AP can successfully ping each other (no packet loss experienced)

I am using Ubuntu 14.04 with Linux 3.13.0 which is within the suggested version range. There were no issues had with any step in the installation phase until this final issue.

Any help would be greatly appreciated!!!

closed 2018-03-06 by struggletown

athpgf, 2018-04-29:
Hi @struggletown,

I am facing the exact same situation. With the same version as you, everything seems fine until I try log_to_file, where I get nothing at all. I am using a TP-Link AP (unencrypted) and a laptop with the IWL5300 as the client, and the AP is sending at HT. So, all the requirements seem to be fulfilled.

Did you get to solve the problem?  Could it be related to the connection mode AP-client, as they kind of suggest in FAQ 10 that it may only work in injection mode with 2 laptops with the IWL5300?

Thanks in advance!

struggletown, 2018-04-29:
Hey!
I did find a fix - it turned out I had my drivers installed in the wrong order. To fix, all I had to do was re-run the commands 
_sudo modprobe -r iwlwifi mac80211_ and _sudo modprobe iwlwifi connector_log=0x1_ 
each time before I connected to the AP. My laptop doesnt seem to hold the driver installation order between reboots, so this process has to be done each time I want to collect, but I'm having success! 
Hope this helps you