dhalperi/linux-80211n-csitool-supplementary #371: Problem getting AoA from CSI data

minseuk, 2018-04-04:
Hello. I'm a student studying Wi-Fi indoor localization using CSI packets.
I got CSI packets with following csitool and got Unwrapped CSI phases.
After ToF sanitization, I can obtain AoA values, but the result is different from actual angle in the environment. I did experiment on 5.745GHz and 3cm antenna distance.

Acutally, I studied this csitool for a long time, and I think my analysis algorithm including sanitization and MUSIC is correct. I guess the problem is caused from CSI data.

For example, 
this is a data for 60 degree, but it shows the AoA is 30 degree (have to add -90 degree in the pseudospectrum graph)
 
[image]

[image]

Is there any issue such as 'phase reverse' or 'x+90 degree x-90 degree' from your experience?
I don't know if it was coincidence, but when i added Pi for certain antenna phase, I got almost exact result. I expect your response.

SusieYonng, 2018-06-01:
Maybe your ground truth is 30 degrees?

minseuk, 2018-06-01:
Hello SusieYonng.  For this example, the phase difference from each antenna is about 0.35*2pi. I did an experiment in 5.745GHz and the 3cm antenna distance, so mathematically It shows asin(0.35/0.5745) = 37 degree. But the ground truth was 60 degree.

soltanaghaei, 2018-07-30:
There is a constant LO offset between the three antennas that need to be manually removed before estimating AoA. Please check ArrayTrack paper for more information and the calibration process.

minseuk, 2018-07-31:
@laxima Hello laxima. I'm sorry for late answer. My email is kms4105@kaist.ac.kr  Thank you.

minseuk, 2018-07-31:
@soltanaghaei Hello soltanaghaei. Thank you for your advice. I found ArrayTrack paper and know how to solve the offset problem. It was represented in 'implementation' part of that paper.

titinho11, 2019-05-23:
Hello Minseuk,

I am also a student working on wifi based indoor localisation system, and i am also facing inconsistant angle of arrival results. Does  Arraytrack's calibration make it work for you ?

zwlovefish, 2019-07-24:
@minseuk hello, i am also dealing with localization system,Could you send me your matlab code ？please help me. thank you

zwlovefish, 2019-07-24:
@minseuk my email is zjw1149610059@gmail.com