dhalperi/linux-80211n-csitool-supplementary #377: CSI amplitude not stable

Statisticss, 2018-05-12:
Hi everyone,

When I was plotting the amplitude of CSI, I found that it seems like there are two levels, as the figure shows below:
[image]

I was wondering why it looks like this.., I mean, I sent all the packets in one channel within a very short time, and the Tx and Rx are unmoved. Does the transmitter change its power? But I thought it may be fixed to 15 dBm, right? Has anyone had the similar situation?

@dhalperi @dpward @asifhanif1122 @DiDomenicoSimone @linkin8834

linkin8834, 2018-05-12:
Hi,

I think that I see this behaviour from time to time. Yes the Txpower is 15dBm but your RX AGC could behave differently for different packets... You might try to plot the scaled version using get_scale_csi (or something similar to this). Also, take a look into the noise level / SNR / RSSI.

Statisticss, 2018-05-13:
Thanks for your quick reply! @linkin8834 

But actually, this is the result after using the get_scale_csi. I also checked the AGC values, they are all the same (32). The RSSI value is either 37 or 38, but they are not matching the figure, I mean, those packets with RSSI 37 doesn't necessarily the lower lines in the figure. The data was collected in monitor mode so the noise level is unavailable and set to -92 in get_scale_csi by default. BTW, I used only 1 Tx and 1 Rx antenna for testing. And I did several experiments, all of them have the similar situation... 

Do you have any suggestions about that? Thanks...

laxima, 2018-07-03:
Hello @Statisticss, 

The CSI should be reported only when you are in HT mode which means that the receiver will use all 3 antennas to receive it. using only one antenna at rx..how did you collect your data?

Statisticss, 2018-07-03:
@chukku Thanks for your confirmation!
@laxima  In fact, HT mode doesn't necessarily mean multiple antennas... Please see the MCS table in FAQ 1 for reference

laxima, 2018-07-03:
Hello @Statisticss,
Yes, I know that , but I am pretty sure that the rx will use all the three antennas. Btw , what rate did you use ?

laxima, 2018-07-11:
HI @Statisticss ,
You mentioned you use one antenna at RX, how did you do that? Can you please share and also what rate did you use.

Statisticss, 2018-07-11:
Hi @laxima , sorry for the delay.
To use one antenna at Rx, you may want to check #34 
I just used 0x4101 as the Tx rate

laxima, 2018-07-12:
@Statisticss , thanks for your reply.

laxima, 2018-07-15:
@ are you able to vary/control tx power?

Statisticss, 2018-07-20:
@laxima No I can't

Statisticss, 2018-07-25:
@linkin8834 Thanks for your suggestion! But I'm wondering that if a router is necessary or not? (I'm currently using laptops with external antennas, and cannot get good results)
And about the Tx power changing, I don't quite get it... Do you mean that I can change Tx power with another WiFi card, and then change back to 5300? Because even if I can change Tx power with other WiFi cards, it doesn't make sense since other cards cannot get CSI

linkin8834, 2018-07-25:
@Statisticss If you really want to change TxPower, you need to do it at the transmitter side. To change the TxPower at the transmitter side, you are not allowed to use 5300 as the transmitter since the firmware disallows you to do so. 

Therefore, you can use a laptop equipped with another WiFi card and then run hostapd to start an AP. In this case, you should be able to change the TxPower of your AP. Your receiver or client is still a 5300 laptop. As long as the packet from the AP is 802.11n, you should be able to get the CSIs even if the transmitter is non-5300. You can check their FAQ for more details.

Are you experimenting with two laptops in monitor/injection mode? If so, I can get good results all the time. I am using internal antennas for my experiments. In this aspect, it can be problem with your external antennas. Are your external antennas designed for 5GHz? 

Also, are you showing the amplitude of raw CSIs in the figure? Is it in linear scale or dB scale?

Statisticss, 2018-08-08:
@linkin8834 Sorry for the delay, I was traveling these days.
Thanks for your answer! I got your point about changing the Tx power.
Yes, I'm currently using two laptops in monitor/injection mode, and the picture I posted is the raw CSI data without any processing and it is in linear scale. It could be a problem of my external antennas, thanks for your suggestion! I will check the antennas soon.