dhalperi/linux-80211n-csitool-supplementary #385: Will this tool still work if I use an adapter to PCI-Express x1?

Azymptote, 2018-06-09:
Looking into buying a NIC for this project, however I do not have a PCI mini slot on my motherboard. Would this work? Or would be have to be customized aswell...

Sunsoar 450Mbps WiFi Wireless PCI-Express x1 Adapter Desktop Card for Intel 5300 Ch

Tamzeed, 2018-06-11:
I have used  adapter to PCI-Express  and got the CSI. So I dont think you will have problem.

h11128, 2020-01-30:
> I have used adapter to PCI-Express and got the CSI. So I dont think you will have problem.

Hi, which adapter you were you used? I encounter the same problem

iftolito, 2020-02-01:
Y

> I have used adapter to PCI-Express and got the CSI. So I dont think you will have problem.

Hi tamzeed i'm using adapter to PCI-Express and i dont got de CSI, my file CSI.DAT is empty can you help me please,

h11128, 2020-02-22:
@iftolito I am using [a mini PCI-E to PCI-e 1X converter](https://www.amazon.com/Fenvi-Express-pcie-Wireless-Adapter/dp/B07H5JJZ88/ref=sr_1_4?keywords=mini+PCIe+to+pcie+1x&qid=1580518609&sr=8-4) and I am able to connect to the internet. After 1 week's configuration, I can get the CSI under monitor mode. And based on my current understanding of the tool, if the NIC card can work normally then it's the problem of configuration that causes you don't get CSI. You will probably need to check FAQ 5 very carefully.