dhalperi/linux-80211n-csitool-supplementary #389: Exploit Phase & Phase Difference

OnArcher, 2018-06-16:
Hi all,

I am using the CSI-Tool for activity recognition. I retrieved the CSI from the "get_scaled_csi", and I calculated the CSI amplitude and phase by Matlab functions, abs(CSI) and angle(CSI).

The CSI amplitude is OK, but the CSI phase is different than I expected.
This is my Raw Phase and Unwrapped Phase.

[image]

[image]

However, I think that the Raw Phase should be Zig-Zag, and the Unwrapped Phase should be almost straight line like this.

[image]

Anybody can help to get fix my phase...

I used 5 GHz...

Thank You.

Tamzeed, 2018-06-20:
Can you make the same plots without any gesture or change in the environment?

ASDF-1, 2018-07-18:
I have the same problem. I don`t know how to use phase information to analyse because it is in a mess.