dhalperi/linux-80211n-csitool-supplementary #393: the 'db' function belongs to the signal package from Octave Forge but has not yet been implemented.

chinmoy009, 2018-06-29:
I am using octave version 4.2.2 and was trying to execute "csi = get_scaled_csi(csi_entry);  " command which gave me a warning which states , " the 'db' function belongs to the signal package from Octave Forge but has not yet been implemented."  followed by few errors given below-

" error: 'db' undefined near line 21 column 11
error: called from
    get_total_rss at line 21 column 9
    get_scaled_csi at line 12 column 14 "

It seems that db function does not implemented in octave signal package. Is there any way to make that command work on octave?

mirasarkis, 2018-07-24:
Im getting the same error, knowing that I've been using the tool since more than a year and it's only now that I got this error.

sidharthpanicker, 2018-07-27:
@chinmoy009 @mirasarkis  Did you find a solution? We are facing the same issue. Any luck on the problem?

mirasarkis, 2018-07-27:
I believe that db refers to decibel. The function db, as called in the csi
tool, aims at transforming power ratios into decibel.

So, I just created a function called db that returns10*log(power/1).

Hope this helps.
Mira

Mandar-Kulkarni, 2018-08-01:
Hi @mirasarkis , can you elaborate a little? From what I understand, csi_trace struct contains 29 elements which have 29 objects each with every object being a 3x3 matrix. How to exactly calculate the db?  Is there any equivalent function in Octave?

Danbo3004, 2018-09-04:
@Mandar-Kulkarni based on [this]( https://www.mathworks.com/help/signal/ref/db.html) documentation we can implement a function that returns the decibel value, which is 10*log(x/1) for power.