dhalperi/linux-80211n-csitool-supplementary #399: Failed Installation?

Greasyjoe, 2018-08-06:
I just making sure I understand this portion.

"Connect to an 802.11n access point, by using (for example) the iw and iproute2 utilities (or if NetworkManager is enabled for the wireless interface, using nm-cli or the graphical applet)."

Does this mean connect to my router? My router has an option to be put into AP mode. Should I do that or is regular router mode OK? 

After running the final step
`sudo modprobe iwlwifi connector_log=0x1`

Should I be able to connect? It tries but never actually connects. Is there a way I can verify that the above step has be successful? It appears to build properly from the appropriate 4.2 kernal tag

Thank you for any help in advance.

Greasyjoe, 2018-08-15:
I had the right kernal version, but the wrong OS, 14.04.5 =/= 14.04.4

jyc843, 2018-08-23:
@Greasyjoe 
 Has your problem been solved?

jyc843, 2018-08-23:
How do you resolve it? I have tried many times? But the problem is also? Can you give me some suggestions?

Greasyjoe, 2018-08-27:
So after installing the firmware using 
`sudo modprobe iwlwifi connector_log=0x1`

you should be able to connect to your router if all security settings are off. I had mistakenly installed Ubuntu 14.04.5 instead of 14.04.4. When I reinstalled I instead used [12.04.5-i386](http://releases.ubuntu.com/12.04.5/ubuntu-12.04.5-desktop-i386.iso), and everything worked as expected.

closed 2018-08-27 by Greasyjoe

bigatl22, 2019-06-13:
@Greasyjoe

I'm getting the same build errors. Can you clarify if those went away completely when you switched to  12.04.5-i386? I am using 14.04.4 but I'm wondering if it matters if I use i386 or 64 bit.