dhalperi/linux-80211n-csitool-supplementary #404: Averaging CSI?

smsam75, 2018-08-24:
Hi,

I'm using CSItool for positioning and I need to reduce the dimension of collected CSI for each point.
Currently, I have 500 packets each including 3x3x30 CSI (3 Tx and 3 Rx and 30 subcarriers).
Is it a good idea to take the average the collected CSI from 3 transmitters of the access point? 

Thanks,
Mohsen

AliAwad1994, 2018-09-30:
what router are you using?
if you want to have a 1 by 3 or 2 by 3 CSI you should either buy a router with 1 or 2 transmitting antennas. or you should fix the transmission rate of your current router [if supported by the firmware] to use 1 only transmitting antenna. you should choose a rate according to the 80211n MCS table.

tahmidzbr, 2018-12-04:
Hi Mohsen, No you shouldn't average across all Tx-Rx streams. As the channel response across each stream is quite different.

smsam75, 2019-05-06:
> Hi Mohsen, No you shouldn't average across all Tx-Rx streams. As the channel response across each stream is quite different.

Thanks Tahmid.