dhalperi/linux-80211n-csitool-supplementary #408: Firmware Crash

johnElic, 2018-10-23:
Hi everyone, I have the following error after firmware crash

[  589.805312] iwlwifi 0000:02:00.0: rssis: 36 40 38 noise: -127 agc: 45 antenna_sel: 09 fake_rate_n_flags=0x100
[  589.808181] iwlwifi 0000:02:00.0: Microcode SW error detected.  Restarting 0x2000000.
[  589.808188] iwlwifi 0000:02:00.0: CSR values:
[  589.808191] iwlwifi 0000:02:00.0: (2nd byte of CSR_INT_COALESCING is CSR_INT_PERIODIC_REG)
[  589.808198] iwlwifi 0000:02:00.0:        CSR_HW_IF_CONFIG_REG: 0X00488304
[  589.808204] iwlwifi 0000:02:00.0:          CSR_INT_COALESCING: 0X0000ff40
[  589.808211] iwlwifi 0000:02:00.0:                     CSR_INT: 0X00000000
[  589.808218] iwlwifi 0000:02:00.0:                CSR_INT_MASK: 0X00000000
[  589.808224] iwlwifi 0000:02:00.0:           CSR_FH_INT_STATUS: 0X00000000
[  589.808231] iwlwifi 0000:02:00.0:                 CSR_GPIO_IN: 0X00000000
[  589.808237] iwlwifi 0000:02:00.0:                   CSR_RESET: 0X00000000
[  589.808244] iwlwifi 0000:02:00.0:                CSR_GP_CNTRL: 0X080403c5
[  589.808251] iwlwifi 0000:02:00.0:                  CSR_HW_REV: 0X00000024
[  589.808257] iwlwifi 0000:02:00.0:              CSR_EEPROM_REG: 0X00000000
[  589.808263] iwlwifi 0000:02:00.0:               CSR_EEPROM_GP: 0X90000004
[  589.808270] iwlwifi 0000:02:00.0:              CSR_OTP_GP_REG: 0X00060000
[  589.808276] iwlwifi 0000:02:00.0:                 CSR_GIO_REG: 0X00080044
[  589.808283] iwlwifi 0000:02:00.0:            CSR_GP_UCODE_REG: 0X00004133
[  589.808289] iwlwifi 0000:02:00.0:           CSR_GP_DRIVER_REG: 0X00000000
[  589.808296] iwlwifi 0000:02:00.0:           CSR_UCODE_DRV_GP1: 0X00000000
[  589.808302] iwlwifi 0000:02:00.0:           CSR_UCODE_DRV_GP2: 0X00000000
[  589.808308] iwlwifi 0000:02:00.0:                 CSR_LED_REG: 0X00000060
[  589.808315] iwlwifi 0000:02:00.0:        CSR_DRAM_INT_TBL_REG: 0X880d2953
[  589.808321] iwlwifi 0000:02:00.0:        CSR_GIO_CHICKEN_BITS: 0X27800200
[  589.808328] iwlwifi 0000:02:00.0:             CSR_ANA_PLL_CFG: 0X00880300
[  589.808335] iwlwifi 0000:02:00.0:           CSR_HW_REV_WA_REG: 0X0001001a
[  589.808341] iwlwifi 0000:02:00.0:        CSR_DBG_HPET_MEM_REG: 0Xffff0000
[  589.808344] iwlwifi 0000:02:00.0: FH register values:
[  589.808362] iwlwifi 0000:02:00.0:         FH_RSCSR_CHNL0_STTS_WPTR_REG: 0X1e799f00
[  589.808380] iwlwifi 0000:02:00.0:        FH_RSCSR_CHNL0_RBDCB_BASE_REG: 0X0210b3d0
[  589.808397] iwlwifi 0000:02:00.0:                  FH_RSCSR_CHNL0_WPTR: 0X000000a0
[  589.808414] iwlwifi 0000:02:00.0:         FH_MEM_RCSR_CHNL0_CONFIG_REG: 0X00801114
[  589.808432] iwlwifi 0000:02:00.0:          FH_MEM_RSSR_SHARED_CTRL_REG: 0X000000fc
[  589.808449] iwlwifi 0000:02:00.0:            FH_MEM_RSSR_RX_STATUS_REG: 0X02130000
[  589.808466] iwlwifi 0000:02:00.0:    FH_MEM_RSSR_RX_ENABLE_ERR_IRQ2DRV: 0X00000000
[  589.808484] iwlwifi 0000:02:00.0:                FH_TSSR_TX_STATUS_REG: 0X07ff0001
[  589.808502] iwlwifi 0000:02:00.0:                 FH_TSSR_TX_ERROR_REG: 0X00000000
[  589.808505] iwlwifi 0000:02:00.0: Loaded firmware version: 8.24.2.12
[  589.808652] iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
[  589.808655] iwlwifi 0000:02:00.0: Status: 0x0000004C, count: 5
[  589.808659] iwlwifi 0000:02:00.0: 0x0000000C | FH_ERROR                    
[  589.808662] iwlwifi 0000:02:00.0: 0x000006E4 | uPc
[  589.808665] iwlwifi 0000:02:00.0: 0x000005AA | branchlink1
[  589.808668] iwlwifi 0000:02:00.0: 0x00000774 | branchlink2
[  589.808671] iwlwifi 0000:02:00.0: 0x000008B2 | interruptlink1
[  589.808674] iwlwifi 0000:02:00.0: 0x0001C6BA | interruptlink2
[  589.808677] iwlwifi 0000:02:00.0: 0x00000020 | data1
[  589.808680] iwlwifi 0000:02:00.0: 0x02130000 | data2
[  589.808683] iwlwifi 0000:02:00.0: 0x000082AC | line
[  589.808686] iwlwifi 0000:02:00.0: 0x404017F8 | beacon time
[  589.808689] iwlwifi 0000:02:00.0: 0x01930808 | tsf low
[  589.808692] iwlwifi 0000:02:00.0: 0x00000000 | tsf hi
[  589.808695] iwlwifi 0000:02:00.0: 0x00000000 | time gp1
[  589.808698] iwlwifi 0000:02:00.0: 0x01A2DCDD | time gp2
[  589.808701] iwlwifi 0000:02:00.0: 0x00000000 | time gp3
[  589.808704] iwlwifi 0000:02:00.0: 0x00010818 | uCode version
[  589.808707] iwlwifi 0000:02:00.0: 0x00000000 | hw version
[  589.808710] iwlwifi 0000:02:00.0: 0x00488304 | board version
[  589.808713] iwlwifi 0000:02:00.0: 0x042200F1 | hcmd
[  589.808716] iwlwifi 0000:02:00.0: 0x6FF63000 | isr0
[  589.808719] iwlwifi 0000:02:00.0: 0x01004000 | isr1
[  589.808721] iwlwifi 0000:02:00.0: 0x00000002 | isr2
[  589.808724] iwlwifi 0000:02:00.0: 0x0041FCC2 | isr3
[  589.808727] iwlwifi 0000:02:00.0: 0x10000112 | isr4
[  589.808730] iwlwifi 0000:02:00.0: 0x000082AC | isr_pref
[  589.808733] iwlwifi 0000:02:00.0: 0x00000008 | wait_event
[  589.808737] iwlwifi 0000:02:00.0: 0x0000003C | l2p_control
[  589.808740] iwlwifi 0000:02:00.0: 0x0000003F | l2p_duration
[  589.808743] iwlwifi 0000:02:00.0: 0x00000027 | l2p_mhvalid
[  589.808746] iwlwifi 0000:02:00.0: 0x00000007 | l2p_addr_match
[  589.808749] iwlwifi 0000:02:00.0: 0x00000000 | lmpm_pmg_sel
[  589.808752] iwlwifi 0000:02:00.0: 0x00000000 | timestamp
[  589.808755] iwlwifi 0000:02:00.0: 0x00000200 | flow_handler
[  589.808821] iwlwifi 0000:02:00.0: Start IWL Event Log Dump: display last 20 entries
[  589.808844] iwlwifi 0000:02:00.0: EVT_LOGT:0027450561:0x00002002:1475
[  589.808859] iwlwifi 0000:02:00.0: EVT_LOGT:0027450561:0x000021a0:1475
[  589.808873] iwlwifi 0000:02:00.0: EVT_LOGT:0027450562:0x000021a1:1475
[  589.808887] iwlwifi 0000:02:00.0: EVT_LOGT:0027450563:0x00001f30:1475
[  589.808901] iwlwifi 0000:02:00.0: EVT_LOGT:0027450563:0x000000db:1475
[  589.808915] iwlwifi 0000:02:00.0: EVT_LOGT:0027450564:0x000021a4:1475
[  589.808930] iwlwifi 0000:02:00.0: EVT_LOGT:0027450564:0x000021a5:1475
[  589.808944] iwlwifi 0000:02:00.0: EVT_LOGT:0027450565:0x00002fff:1475
[  589.808958] iwlwifi 0000:02:00.0: EVT_LOGT:0027450566:0x00000000:0334
[  589.808972] iwlwifi 0000:02:00.0: EVT_LOGT:0027450568:0x0000b008:0334
[  589.808986] iwlwifi 0000:02:00.0: EVT_LOGT:0027450568:0x0000b009:0334
[  589.809000] iwlwifi 0000:02:00.0: EVT_LOGT:0027450570:0x0000b011:0334
[  589.809014] iwlwifi 0000:02:00.0: EVT_LOGT:0027450570:0x0000b016:0334
[  589.809028] iwlwifi 0000:02:00.0: EVT_LOGT:0027450571:0x00005001:1475
[  589.809042] iwlwifi 0000:02:00.0: EVT_LOGT:0027450572:0x00005110:1475
[  589.809056] iwlwifi 0000:02:00.0: EVT_LOGT:0027450573:0x00005130:1475
[  589.809071] iwlwifi 0000:02:00.0: EVT_LOGT:0027450573:0x00005131:1475
[  589.809085] iwlwifi 0000:02:00.0: EVT_LOGT:0027450574:0x00005133:1475
[  589.809099] iwlwifi 0000:02:00.0: EVT_LOGT:0027450575:0x000000d7:0123
[  589.809113] iwlwifi 0000:02:00.0: EVT_LOGT:0027450593:0x00000000:0125
[  589.811879] ieee80211 phy0: Hardware restart was requested
[  589.811931] iwlwifi 0000:02:00.0: L1 Disabled - LTR Disabled
[  589.814901] iwlwifi 0000:02:00.0: Radio type=0x0-0x2-0x0
[  589.955480] iwlwifi 0000:02:00.0: L1 Disabled - LTR Disabled
[  589.958474] iwlwifi 0000:02:00.0: Radio type=0x0-0x2-0x0

[+] I have tried to downgrade the kernel and manipulate the iwlwifi parameters, but nothing works ! 
[+] I had removed the following code ( to see the output of firmware ) from 
/linux-80211n-csitool/drivers/net/wireless/iwlwifi/pcie

	/* Error detected by uCode */
	if (inta & CSR_INT_BIT_SW_ERR) {
		IWL_ERR(trans, "Microcode SW error detected. "
			" Restarting 0x%X.\n", inta);
		isr_stats->sw++;
		iwl_pcie_irq_handle_error(trans);
		handled |= CSR_INT_BIT_SW_ERR;
	} 

The error that has been appeared is :
Unhandled INTA bits 0x02000000
iwlwifi : No space in command queue
iwlwifi : Restarting Adapter queue is full
iwlwifi : Error sending REPLY_LEDS_CMD: enqueue_hcmd failed : -28

I'm using injection mode.

@dhalperi @dpward

johnElic, 2018-10-23:
@dhalperi @dpward

johnElic, 2018-10-28:
@dhalperi @dpward

johnElic, 2018-11-04:
Anyone ? @dhalperi @dpward

johnElic, 2018-11-07:
@ozawali I'm wondering if you can help me.

zhangshaohu, 2019-10-29:
I tried this command to solve this issue:
sudo rfkill unblock wifi 
Ref: https://blog.csdn.net/caichao08/article/details/53894510