dhalperi/linux-80211n-csitool-supplementary #411: unwraped phase is not linear

764398256, 2018-11-06:
[image]
The unwraped phase should be linear，but not .why?
hope your help

764398256, 2018-11-13:
> Because of the multipath effect in the wireless channel and temporal phase offset/rotation from non-synchonization between two transceiver. If you manually connect tx and rx antennas with cable and measure the CSI, you should get a linear trend of CSI phase.

Thank you for your answer.
I want to get a more accurate CSI to verify the algorithm. If i manually connect tx and rx antennas with cable of different length and measure the CSI。Can the CSI  accurately calculate the angle？

764398256, 2018-11-13:
@KeNopphon 
Thank you for your answer.
Now，i want to get a more accurate CSI to verify the algorithm. If i manually connect tx and rx antennas with cable of different length and measure the CSI。Can the CSI  accurately calculate the angle？

764398256, 2018-11-13:
@KeNopphon 
thank you very much！

764398256, 2018-11-19:
In order to verify the correctness of the algorithm。If i connect 1 Tx and 3 Rx up through the power divider ，the received 3 signals should be the same。So the estimated angle is zero ideally。Is my idea correct? @KeNopphon

KeNopphon, 2018-11-19:
Ideally, if there is no phase offset/rotation and the considering the power divider is ideal(no reflection signal between ports and have the same gain) at all, you may receive the same signal. I am not sure what do u mean by the estimated angle but I assume it meant the phase difference of CSI between 2 ports which according to the previous sentence assumption could produce the zero CSI phase difference.

However, if you want to verify the correctness of the CSI phase, I think it is better to compare with the result from VNA, SDR instead