dhalperi/linux-80211n-csitool-supplementary #415: AGC

lax7, 2019-01-10:
Does every RF chain have its own AGC? @dpward @dhalperi,

q760019473, 2019-03-14:
I have such a question. Can you leave an email for communication?

lax7, 2019-03-28:
Sorry for late reply, Give me your Wechat and I will add you.

q760019473, 2019-03-28:
OK  thankyou  Wechat： q1440606348

NickiXD, 2019-05-15:
Hi,
maybe there is someone who can help me?  Erverybody talks about the AGC value to scalling the CSI. I read the Datasheet of the Atheros chipset but i could't find any description about this amplifier value. So my questions are: 
Why is this value so importent to process my CSI data?
Where is this value exactly come from or where can I read about that? 
I'd be very grateful if anyone could help me or point me to a source.
Thank you :)