dhalperi/linux-80211n-csitool-supplementary #418: Cannot get 3X3X30 CSI by using 3-antenna AP

lee-buaa, 2019-03-22:
I have used a TP-Link with 3 antenna as AP, a laptop with Intel 5300 as client. After following installation guide, I can only get 1X3X30 CSI matrix.  And if I use 0x1c910 to set transmitter rate of AP through my laptop, I cannot ping the AP. So how should I set to get 3X3X30 CSI matrix?

steven520142, 2019-03-26:
I have the same problem. Sometimes I receive 1x3x30 CSI matrix but sometimes I receive 2x3x30 CSI matrix. Does someone know what's going on?

lee-buaa, 2019-03-26:
According to FAQ#2, we need to set some parameters on Tx through client (Rx). But the problem is I cannot ping the AP if I set them. Do you try this method?

lee-buaa, 2019-04-02:
To solve this problem, you can use external antenna.

closed 2019-04-02 by lee-buaa

chenlk000, 2020-04-27:
I meet this problem too, setting for parameters is not work for me, just get 1x3x30 and 2x3x30. So I set another 5300 in inject mode and receive the csi in monitor mode, after setting parameters shows in FAQ#2 then I got 3x3x30 csi matrix sucessfully.